ucb-bar/gemmini #16: System stuck on mvin (FPGA platform)

KireinaHoro, 2020-03-23:
I'm trying to run the `baremetalC` tests (commit https://github.com/ucb-bar/gemmini-rocc-tests/tree/3da2662f551556ce736bb215eca88af34ee13cb4) on an FPGA platform (Xilinx ZCU102), but the Rocket Chip system seems to stall on the __second__ `mvin` instruction: the first will commit successfully, but the second one will get stuck.  The whole system seems to stall as OpenOCD fails to halt the hart in this case.

My setup is `WithNBigCores(1)` plus `DefaultGemminiConfig`.  I've enabled the [inclusive cache from SiFive](https://github.com/sifive/block-inclusivecache-sifive) for testing.  Not sure if this would be a problem as to my knowledge theoretically Gemmini accesses DRAM through TileLink and this shouldn't be a problem provided that the CPU code is already running, but I'd be willing to test if this would possibly cause problems.

The software runs in S-mode (OpenSBI runs in M-mode).  I've enabled the XS bit in M-mode:

```c
        /* Enable Gemmini */
        if (misa_extension('X'))
                csr_set(CSR_MSTATUS, MSTATUS_XS & (MSTATUS_XS >> 1));
``` 

I've tried with and without VM (identity mapping when VM is on); both setup gives the same results.  Would love any form of assistance, thanks.

hngenc, 2020-04-12:
Unfortunately, we have not tried running on an FPGA before; only on Verilator, VCS, or FireSim simulations. We also taped-out some ASICs, but we have not had the opportunity to test on those yet.

Does your config work in Verilator or VCS simulations? Does it only fail when you try to move it onto the FPGA?

KireinaHoro, 2020-04-13:
I've got it working with the recently-merged change of padding logic in hardware.  The commit mentioned in this issue had problems inside the StreamReader implementation such that a mismatch between decoded transfer length and actual transfer length triggered an assertion.  The assertion (confirmed by simulation), resulted in RoCC's busy signal never being de-asserted, thus hanging the whole system.

With the master commit, I've got the tests up and running.  Infrequent hangs are still observed with no obvious patterns, but the software tests will finish most of the time.  I will report if I pinpoint the actual problem.  Closing for now.

closed 2020-04-13 by KireinaHoro