ucb-bar/gemmini #17: Software test resnet50

xiongwu12345, 2020-04-02:
When I build and run，I encounter this error
‘’‘spike --extension=gemmini resnet50-baremetal 
Gemmini extension configured with:
    dim = 16
Killed’‘’
So what should I do ？
Thanks a lot！

xiongwu12345, 2020-04-09:
Does anyone like to help me?Thanks a lots!
It is a emergency!

hngenc, 2020-04-12:
Which commits are you on for `gemmini` and `chipyard`?

There is a known issue where the Spike simulation repo may be out-of-sync with the Gemmini repo, unfortunately: https://github.com/ucb-bar/chipyard/pull/509

If that is your problem, you can fix that by running these commands:

```
cd toolchains/esp-tools/riscv-isa-sim
git checkout 3c930b403140ac0cd1fc5578a21998f6875459c6
cd build
make
make install
```

But before running that, you can tell us which commits you're on so that we can double-check whether you need to do this.

closed 2022-01-21 by hngenc