ucb-bar/gemmini #20: run other neural networks on gemmini besides mobilenet and resnet50

zhou-zhiyuan, 2020-05-12:
how can I run other neural networks on gemmini?
For example(mobilenet) ,I find mobilenet.c and mobilenet_params.h.
Can you provide some tools or give some guidance on generating these two files according to different neural networks？ 
Thanks！

pranav-prakash, 2020-05-12:
The mobilenet and resnet50 examples in gemmini-rocc-tests were manually crafted after performing post-training quantization on the two networks. 

For running arbitrary models, you might find it easier to use [onnxruntime-riscv](https://github.com/pranav-prakash/onnxruntime-riscv/) which we've been developing as a tool to schedule arbitrary onnx models on gemmini. The relevant gemmini specific documentation can be found in the [systolic_runner](https://github.com/pranav-prakash/onnxruntime-riscv/tree/systolic/systolic_runner) folder, and contains tools for automatically performing post-training quantization on any floating point onnx model as well as a sample runner that can be used with any imagenet model (e.g. Googlenet, squeezenet, etc.).

Running non-imagenet models is also feasible (one of our teams was experimenting with running BERT), it's just a matter of writing the appropriate runner. See [custom_op_test_src/runner.cpp](https://github.com/pranav-prakash/onnxruntime-riscv/blob/systolic/systolic_runner/halide_interop/custom_op_test_src/runner.cpp) for a simple example of a runner for a graph that takes 2 inputs and produces 1 output. You can refer to Microsoft's own documentation on the details of the c++ api.

zhou-zhiyuan, 2020-05-13:
Thanks for your answer. I will take look at onnxruntime-riscv and push issues under that project if meet with other questions.

closed 2020-05-13 by zhou-zhiyuan