ucb-bar/gemmini #21: std::out_of_range error

zzoon91, 2020-05-19:
Hi, i'm a newbie in chipyard and interested gemmini project

I encountered error while following the documentations and READMEs
When I run `spike --extension=gemmini mvin_mvout-baremetal`
it says
```
Gemmini extension configured with:
    dim = 16
terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 2097152) >= this->size() (which is 2097152)
```

I configured my repository with
https://chipyard.readthedocs.io/en/latest/Chipyard-Basics/Initial-Repo-Setup.html 
(I ran the build script with esp-tools option)
Then I followed the "Build and Run Gemmini Tests" section in the README of this repository

My environment is Ubuntu 18.04.3

I appreciate your help in advance :)

pranav-prakash, 2020-05-19:
I've encountered this error before when the version of Spike doesn't match with the instruction encoding used in rocc-tests. If you're using Chipyard's spike then I'm not sure it has the latest changes. (@hngenc might be a good idea to bump chipyard).

Perhaps try using spike from `master` branch from https://github.com/ucb-bar/esp-isa-sim
Also you might need to update the gemmini-rocc-tests submodule to the latest (https://github.com/ucb-bar/gemmini-rocc-tests) so that it uses a `gemmini_params.h` matching what Spike expects.

zzoon91, 2020-05-20:
It worked!
Thanks

closed 2020-05-20 by zzoon91

hngenc, 2020-05-20:
Sorry about that. This has been fixed in `dev`. Chipyard master should be bumped soon, and this should be resolved then.