ucb-bar/gemmini #25: Gemmini tests with spike

pbert519, 2020-06-10:
I use chipyard 1.3.0 with gemmini caaf781 and https://github.com/ucb-bar/esp-isa-sim/commit/13384cac1e54828200067ff890f564a505a4ebb3. When running the gemmini baremetal tests, e.g.
`spike --extension=gemmini build/bareMetalC/mvin_mvout_acc-baremetal` 
the tests fails. The Matrix output is null for all entries.
I also tried https://github.com/ucb-bar/esp-isa-sim/commit/2e9696576f3f19f2fe427a681cde82a326a23a04 with the same result. 
RISCV toolchain is the esp-tools variant.

Can you give some hint for a working riscv-isa-sim and gemmini version combination?

The output for the mvin_mvout_acc-baremetal test is attached:
[gemmini.log](https://github.com/ucb-bar/gemmini/files/4757751/gemmini.log)

A other question, is it possible to define a different array dimension for spike?

hngenc, 2020-06-19:
To change the systolic array dimension in Spike, you can change this [line](https://github.com/ucb-bar/esp-isa-sim/blob/2e9696576f3f19f2fe427a681cde82a326a23a04/gemmini/gemmini_params.h#L7).

Chipyard v1.3.0 should have worked... Do you know if any of the other tests also failed? Also, did you build Gemmini with your own configuration (like a different systolic array size) before running the Spike tests?

I'll clone and run Chipyard 1.3.0 myself just to be sure, and I'll get back to you.

pbert519, 2020-06-19:
Thanks for looking into it and the tip about the spike configuration.

I didn't rebuild gemmini with a different configuration and the gemmini_params.h file has the default values. I compared the gemmini_params.h in the gemmini_rocc_tests and in riscv-isa-sim, they have the same parameters.

All gemmini-rocc-test failed for me as baremetal test with spike except the mvin_scale and tiled_matmul_cpu. 

In the meantime i also tested the linux version of the tests with firemarshal - gemmini-tests.json - and spike. I attached the log, warning: big size:
[uartlog.txt](https://github.com/ucb-bar/gemmini/files/4805104/uartlog.txt)
The problem is the same: All output matrix entries are 0.
I used the newer ucb-bar/esp-isa-sim@2e96965 spike version for this.

Btw. the verilator sim is working for the baremetal gemmini tests.

hngenc, 2020-06-20:
Can you try running these commands?

```
cd chipyard/
cp generators/gemmini/software/gemmini-rocc-tests/include/gemmini_params.h toolchains/esp-tools/riscv-isa-sim/gemmini/gemmini_params.h
cd toolchains/esp-tools/riscv-isa-sim/build/
make && make install
```

And then trying to run the -baremetal tests with Spike again?

pbert519, 2020-06-20:
Now it's working. Thank you!

closed 2020-06-20 by pbert519

hngenc, 2020-06-20:
No problem. It turns out that the `gemmini_params.h` file in `esp-tools` and `gemmini-rocc-tests` did not match. We'll fix that in Chipyard master. Thanks for uncovering this bug.

mik300, 2024-12-15:
> Can you try running these commands?
> 
> ```
> cd chipyard/
> cp generators/gemmini/software/gemmini-rocc-tests/include/gemmini_params.h toolchains/esp-tools/riscv-isa-sim/gemmini/gemmini_params.h
> cd toolchains/esp-tools/riscv-isa-sim/build/
> make && make install
> ```
> 
> And then try to run the `-baremetal` tests with Spike again?

The riscv-isa-sim/ directory is empty for me, did I do something wrong during the Chipyard setup?