ucb-bar/gemmini #27: Float datatype

pbert519, 2020-07-31:
I try to build Gemmini with single precision float as datatype.

 Unfortunately i get ta scala.NotImplementedError with the following configuration:
```object GemminiConfigs {
   import Arithmetic.FloatArithmetic._

  //val defaultConfig = GemminiArrayConfig[SInt, SInt](
  val defaultConfig = GemminiArrayConfig[Float, Float](
    tileRows = 1,
    tileColumns = 1,
    meshRows = 4,
    meshColumns = 4,
    //meshRows = 16,
    //meshColumns = 16,
    ld_queue_length = 8,
    st_queue_length = 2,
    ex_queue_length = 8,
    rob_entries = 16,
    sp_banks = 4,
    acc_banks = 1,
    sp_capacity = CapacityInKilobytes(256),
    shifter_banks = 1, // TODO add separate parameters for left and up shifter banks
    dataflow = Dataflow.BOTH,
    acc_capacity = CapacityInKilobytes(64),
    mem_pipeline = 1,
    dma_maxbytes = 64, // TODO get this from cacheblockbytes
    dma_buswidth = 128, // TODO get this from SystemBusKey
    aligned_to = 1,
    //inputType = SInt(8.W),
    //outputType = SInt(19.W),
    //accType = SInt(32.W),
    inputType = Float(8, 24),
    outputType = Float(8, 24),
    accType = Float(8, 24),
    // mvin_scale_args = Some(MvinScaleArguments((t: SInt, u: SInt) => t * u, 0, SInt(8.W))),
    // mvin_scale_acc_args = Some(MvinScaleArguments((t: SInt, u: SInt) => t * u, 0, SInt(8.W))),
    mvin_scale_args = None,
    mvin_scale_acc_args = None,
    // mvin_scale_args = Some(MvinScaleArguments((t: Float, u: Float) => t * u, 0, Float(8, 24))),
    // mvin_scale_acc_args = Some(MvinScaleArguments((t: Float, u: Float) => t * u, 0, Float(8, 24))),
    mvin_scale_shared = false,
    pe_latency = 10
  )
}
```

I use the standard GemminiRocketConfig in Chipyard and `make verilog CONFIG=GemminiRocketConfig` to generate verilog. The output is attached: [log.txt](https://github.com/ucb-bar/gemmini/files/5007831/log.txt)

hngenc, 2020-08-03:
The issue seems to be caused by this line:

https://github.com/ucb-bar/gemmini/blob/master/src/main/scala/gemmini/Arithmetic.scala#L274

We'll add an implementation for that as soon as possible.

In the meantime, you should be able to replace that line as follows, assuming that you don't want to use Gemmini's new max-pooling feature:

```
      override def >(t: Float): Bool = false.B
```

pbert519, 2020-08-03:
That solves the issue, Thank you.

closed 2020-08-03 by pbert519