ucb-bar/gemmini #31: Failed to Compile a fp32 Gemmini

Entropy-xcy, 2020-09-14:
Hi,

The compilation for the following config of gemmini fails:
```
 object GemminiConfigs {
   import Arithmetic.FloatArithmetic._

   val defaultConfig = GemminiArrayConfig[Float, Float](
     tileRows = 4,
     tileColumns = 4,
     meshRows = 4,
     meshColumns = 4,
     ld_queue_length = 8,
     st_queue_length = 2,
     ex_queue_length = 8,
     rob_entries = 16,
     sp_banks = 4,
     acc_banks = 1,
     sp_capacity = CapacityInKilobytes(256),
     shifter_banks = 1, // TODO add separate parameters for left and up shifter banks
     dataflow = Dataflow.BOTH,
     acc_capacity = CapacityInKilobytes(64),
     mem_pipeline = 1,
     dma_maxbytes = 64, // TODO get this from cacheblockbytes
     dma_buswidth = 128, // TODO get this from SystemBusKey
     aligned_to = 1,
     inputType = Float(8, 32),
     outputType = Float(8, 32),
     accType = Float(8, 32),
     mvin_scale_args = None,
     mvin_scale_acc_args = None,
     mvin_scale_shared = false,
     pe_latency = 0
   )
 }
```
The error code is:
```
java.lang.IllegalArgumentException: requirement failed: each SRAM bank must have a power-of-2 rows, to simplify address calculations
```
Is there something mis-configured in my GemminiConfig?

Thanks!

Entropy-xcy, 2020-09-14:
By the way, could you please clarify the Floating point datatype a bit more? Is Float(8, 24) a typical Float32 or is Float(8, 32) a typical Float32?

hunshimowangzhy, 2020-09-18:
I think a 32-bit fp should be Float(8,24). You can find its definition in arithmatic.scala. 
I tried to use floating point before but with no success. The function overload for fp operation is not completely defined in arithmatic.scala. Even after I finished defining those functions for fp, I could not  synthesize it. It seemed that all those fp units need to be synthesized into logic instead of using some floating point dsp, which was taking a hell lot of time and I just gave up. I used fixed-point instead which is better supported by chisel.
If you have any success please let me know because I am also interested in a floating point implementation.

hngenc, 2021-03-28:
Sorry for the late reply. Floating-point types are working now in the `dev` branch. You can look for commonly-used floating point configurations in `ConfigsFP.scala`, where `FP32`, `BF16` and other configurations are defined.

Also, if you use the `dev` branch, then be sure to check the `CHIPYARD.hash` and `SPIKE.hash` files, which will let you know which Chipyard and Spike commits you need to be on to match your Gemmini version.

closed 2022-01-21 by hngenc