ucb-bar/gemmini #35: how to run the unit test of MeshWithDelaysTester

xiongwu12345, 2020-11-08:
hello all guys，I want to run the mesh unit test file with the command  "**testOnly gemmini.MeshWithDelaysTester**" in the gemmini directory with sbt run first.But I get this error :
"
[error] 
[error]   last tree to typer: This(class StreamReadRequest)
[error]        tree position: line 24 of /home/xiongwu/chipyard/chipyard/generators/gemmini/src/main/scala/gemmini/DMA.scala
[error]             tree tpe: StreamReadRequest.this.type
[error]               symbol: class StreamReadRequest in package gemmini
[error]    symbol definition: class StreamReadRequest[U <: chisel3.Data] extends AnyRef (a ClassSymbol)
[error]       symbol package: gemmini
[error]        symbol owners: class StreamReadRequest
[error]            call site: method cloneType in class StreamReadRequest in package gemmini
[error] 
[error] == Source file context for tree position ==
[error] 
[error]     21   val len = UInt(16.W) // TODO magic number
[error]     22   val cmd_id = UInt(8.W) // TODO magic number
[error]     23 
[error]     24   override def cloneType: StreamReadRequest.this.type = new StreamReadRequest(spad_rows, acc_rows, mvin_scale_t_bits).asInstanceOf[this.type]
[error]     25 }
[error]     26 
[error]     27 class StreamReadResponse[U <: Data](spadWidth: Int, accWidth: Int, spad_rows: Int, acc_rows: Int, aligned_to: Int, mvin_scale_t_bits: Int)
[error] Total time: 1 s, completed Nov 8, 2020, 10:50:47 AM
"
Did anyone have run the unit test successfully?Can u give me some advice?
Thank you so much!!!

hngenc, 2020-11-15:
The unit tests are very much out-of-date; we should probably just delete them by this point.

We use the tests in `software/gemmini-rocc-tests` instead.

closed 2020-11-15 by hngenc

xiongwu12345, 2020-11-19:
Oh,I understand,thank you!

Closed #35.