ucb-bar/gemmini #71: spike mobile_net  failed

hyy313, 2021-03-04:
[image]

follow#25, i can spike mvin_mvout, but when i spike mobile_net and res_net ,this error happen.

parameters problem?

hyy313, 2021-03-04:
i find the problem, when i run my case, find ths rs1>>32 error

[image]

[image]

65548>>32 is not 1006632960

Zeldax64, 2021-03-27:
Use chipyard at version 1.3.0 and not at 1.4.0 (lastest version).

I got the same problem when simulating mobilenet-baremetal. It seems like a problem with rs1 value in a custom3 instruction when used to configure Gemmini. According to the [documentation](https://github.com/ucb-bar/gemmini#config_ex-configures-the-execute-pipeline), the new_acc_shift variable should be in bits [63:32] of rs1. However, the value in this field was 0x3c000000, which is much bigger than sizeof(acc_t)*8. I believe the problem is related to the preprocessor macros used to encode gemmini operations in [gemmini.h](https://github.com/ucb-bar/gemmini-rocc-tests/blob/2802ca406323ef511e1c2939387144a47c250638/include/gemmini.h#L244).

hngenc, 2021-03-28:
Yes, unfortunately, it seems the preprocessor macros were broke. I would recommend switching over to the `dev` branch of Gemmini, where these issues have been resolved (we probably won't bump `master` till Chipyard gets another major release).

If you do want to use the `dev` branch, then you can check the `CHIPYARD.hash` and `SPIKE.hash` files, which will tell you exactly which Chipyard and which Spike commit you need to be on.

AaronJing, 2021-04-06:
I switched to the `dev` branch and checkout to the commit from hash files. I recompiled and installed the spike. When I was running the test, I received the following output. 
[image]
Is this "CORRECT" output? Or I missed something after checkout to the commit you mentioned above.

hngenc, 2021-04-06:
Are you sure you remembered to `git submodule update` after pulling `dev`?

```
spike --extension=gemmini build/imagenet/mobilenet-baremetal
```

AaronJing, 2021-04-07:
@hngenc 
Oh, I did not see there are submodules. Now the problem fixed. Thanks.

hngenc, 2021-04-07:
@AaronJing

Great, thanks for the update!

closed 2022-01-21 by hngenc