ucb-bar/gemmini #105: broke hardware running

alexhilaev, 2021-04-12:
hello, i`m trying to run project on real hardware(xilinx) with example config:
```
class RV64GEMMINIConfig extends Config(
  new chipyard.harness.WithSimAXIMem ++
  new gemmini.DefaultGemminiConfig ++                            // use Gemmini systolic array GEMM accelerator
  new freechips.rocketchip.subsystem.WithNBigCores(1) ++
  new chipyard.config.AbstractConfig
)
```

baremetal isa tests runs ok, but baremetal gemmini such as mvin-mvout hangs gemmini`s pipeline
```
Assertion failed: pipeline stall
    at ROB.scala:246 assert(cycles_since_issue < 10000.U, "pipeline stall")
```

Is any hints i don't know?

hngenc, 2021-04-15:
We haven't tried to run on real FPGAs before (although I think a user did manage to do that once). We've only been running with Firesim and VCS/Verilator.

Does your `RV64GEMMINIConfig` configuration pass the baremetal tests with Verilator?

alexhilaev, 2021-05-24:
No, it is not. If i try to add simaxi instead of tlram, verilator freezes on start.