ucb-bar/gemmini #110: Alternative SoC Configs doesn't compile

sammhho, 2021-04-21:
Hi,
I tried following the [alternative-soc-configs](https://chipyard.readthedocs.io/en/latest/Generators/Gemmini.html#alternative-soc-configs) part of Gemmini documentation to compile,
and got this error:

> [info] running freechips.rocketchip.system.Generator --target-dir /home/samh/repos/github.com/ucb-bar/chipyard/sims/verilator/generated-src/freechips.rocketchip.system.TestHarness.EfxGemminiAcceleratorConfig --name freechips.rocketchip.system.TestHarness.EfxGemminiAcceleratorConfig --top-module freechips.rocketchip.system.TestHarness --legacy-configs gemmini:EfxGemminiAcceleratorConfig
[31m------------------------------------------------------------------------------
Error: Unknown option --legacy-configs
Try --help for more information.
------------------------------------------------------------------------------[0m

Are these "Example TopLevel Configs" in `chipyard/generators/gemmini/src/main/scala/gemmini/Configs.scala` deprecated?
If so, what would be an equivalent config to `GemminiAcceleratorConfig`, e.g. a small host with a Gemmini, in terms of the configs in `chipyard/generators/chipyard/src/main/scala/config/RocketConfigs.scala`?
Is that similar to the `TinyRocketConfig` with `new freechips.rocketchip.subsystem.With1TinyCore` set?
Is the inclusion of RoCC automatically configured? coz there seems no explicit mentioning in the RocketConfigs file but I'm sure some other configs are using, say HwachaRocketConfig.

hngenc, 2021-04-21:
The alternative configs were indeed deprecated (and removed in #108). We'll update the documentation to remove references to them.

I usually build [this default config in RocketConfigs.scala](https://github.com/ucb-bar/chipyard/blob/master/generators/chipyard/src/main/scala/config/RocketConfigs.scala#L31-L34):

```
make CONFIG=GemminiRocketConfig
```

It comes with a small in-order Rocket CPU hooked up to Gemmini through the RoCC interface, which seems to be exactly what you want.

I'm not sure why you don't see a Gemmini config in your own RocketConfigs.scala file. Perhaps you need to bump up to `master`?

sammhho, 2021-04-21:
@hngenc Thx for the quick response. Maybe I didn't pull for a while, I really should.
To clarify, I did see and have used the `GemminiRocketConfig` before.
I was just trying to learn and use a customized GemminiConfig instead of `DefaultGemminiConfig`,
and then at the same time maybe try out some smaller config of Rocket. 
Coz in the `GemminiRocketConfig` it used `freechips.rocketchip.subsystem.WithNBigCores(1)`, so i just guess it's larger than the deprecated `WithGemminiHostMiniCore` or the `freechips.rocketchip.subsystem.With1TinyCore`.
I'm trying to evalute Gemmini's performance on a small FPGA.
BTW do we know if the comparison against NVDLA [here](https://charleshong3.github.io/projects/nvdla_v_gemmini.pdf) is a fair one?

sammhho, 2021-04-21:
Found the answers [here](https://github.com/ucb-bar/chipyard/issues/858#issuecomment-821574869), closing issue.

closed 2021-04-21 by sammhho

hngenc, 2021-04-21:
With regards to the NVDLA question:

We compared against an NVDLA config with the same number of MACs per cycle. Changing the dataflow to WS does **not** halve the number of MACs compared to BOTH. BOTH just adds some extra muxes around the PEs, but it doesn't change the number of multipliers or adders.

sammhho, 2021-04-21:
> With regards to the NVDLA question:
>
> We compared against an NVDLA config with the same number of MACs per cycle. Changing the dataflow to WS does **not** halve the number of MACs compared to BOTH. [...]

I see. Will try if this holds true for different synth tools, perhaps the one I try was bad. Thx.

sammhho, 2021-04-29:
@hngenc Just FYI, by setting InputType = 16.W (to force using DSP instead of LUT on FPGAs), 
with PE_256 set as top in Vivado, number of DSPs used is 3;
For the OS path, although in RTL it's 1 mult, 2 adds, Vivado chose to replicate the paths and used 2 DSPs;
For the WS path, the expected operator sharing did happen so it's 1 DSP;
Operator sharing did not happen between OS and WS paths, so it ended up using 3 macs;
Our in-house synthesis tool for FPGA reports 2 macs as well, not being able to share between OS and WS;
It'd be good if results from Synopsys DC can be found somewhere;

hngenc, 2021-04-29:
Huh, that's interesting. I'll check with our own synthesis tools to see if MAC-sharing happens. Thanks for the heads-up.