ucb-bar/gemmini #117: Error when compiling gemmini

Z-KN, 2021-05-07:
when I try to compile gemmini to get .v files,I try the following command but have the following errors, which seems about syntax.

username0@Ubuntu:~/Chipyard/chipyard/generators/gemmini$ sbt "test:runMain gemmini.Laucher all"
[info] welcome to sbt 1.3.13 (Ubuntu Java 11.0.10)
[info] loading settings for project gemmini-build from plugins.sbt ...
[info] loading project definition from /home/username0/Chipyard/chipyard/generators/gemmini/project
[info] loading settings for project gemmini from build.sbt ...
[info] set current project to gemmini (in build file:/home/username0/Chipyard/chipyard/generators/gemmini/)
[info] Compiling 43 Scala sources to /home/username0/Chipyard/chipyard/generators/gemmini/target/scala-2.12/classes ...
[error] /home/username0/Chipyard/chipyard/generators/gemmini/src/main/scala/gemmini/AccumulatorMem.scala:79:39: value rdata is not a member of chisel3.Bundle
[error]   val w_sum = VecInit((RegNext(mem.io.rdata) zip wdata_buf).map { case (rv, wv) =>
[error]                                       ^
[error] /home/username0/Chipyard/chipyard/generators/gemmini/src/main/scala/gemmini/AccumulatorMem.scala:80:17: value zip is not a member of Any
[error]     VecInit((rv zip wv).map(t => t._1 + t._2))
[error]                 ^
[error] /home/username0/Chipyard/chipyard/generators/gemmini/src/main/scala/gemmini/AccumulatorMem.scala:83:10: value waddr is not a member of chisel3.Bundle
[error]   mem.io.waddr := waddr_buf
[error]          ^
[error] /home/username0/Chipyard/chipyard/generators/gemmini/src/main/scala/gemmini/AccumulatorMem.scala:84:10: value wen is not a member of chisel3.Bundle
[error]   mem.io.wen := w_buf_valid
[error]          ^
[error] /home/username0/Chipyard/chipyard/generators/gemmini/src/main/scala/gemmini/AccumulatorMem.scala:85:10: value wdata is not a member of chisel3.Bundle
[error]   mem.io.wdata := Mux(acc_buf, w_sum, wdata_buf)
[error]          ^
[error] /home/username0/Chipyard/chipyard/generators/gemmini/src/main/scala/gemmini/AccumulatorMem.scala:86:10: value mask is not a member of chisel3.Bundle
[error]   mem.io.mask := mask_buf
[error]          ^
...
[error] 1399 errors found
[error] (Compile / compileIncremental) Compilation failed

Z-KN, 2021-05-07:
I want to generate verilog files of gemmini merely. After modifying 'build.sbt' according to https://stackoverflow.com/questions/58365679/value-is-not-a-member-of-chisel3-bundle, there remain 42 errors.

------
username0@Ubuntu:~/Chipyard/chipyard/generators/gemmini$ sbt compile
[info] welcome to sbt 1.3.13 (Ubuntu Java 11.0.10)
[info] loading settings for project gemmini-build from plugins.sbt ...
[info] loading project definition from /home/username0/Chipyard/chipyard/generators/gemmini/project
[info] loading settings for project gemmini from build.sbt ...
[info] set current project to gemmini (in build file:/home/username0/Chipyard/chipyard/generators/gemmini/)
[info] Executing in batch mode. For better performance use sbt's shell
[info] Compiling 43 Scala sources to /home/username0/Chipyard/chipyard/generators/gemmini/target/scala-2.12/classes ...
[error] /home/username0/Chipyard/chipyard/generators/gemmini/src/main/scala/gemmini/DMA.scala:9:8: not found: object testchipip
[error] import testchipip.TLHelper
[error]        ^
[error] /home/username0/Chipyard/chipyard/generators/gemmini/src/main/scala/gemmini/DMA.scala:113:14: not found: value TLHelper
[error]   val node = TLHelper.makeClientNode(
[error]              ^
[error] /home/username0/Chipyard/chipyard/generators/gemmini/src/main/scala/gemmini/DMA.scala:114:5: value name_= is not a member of gemmini.StreamReaderCore[T,U,V]
[error]     name = "stream-reader", sourceId = IdRange(0, nXacts))
[error]     ^
[error] /home/username0/Chipyard/chipyard/generators/gemmini/src/main/scala/gemmini/DMA.scala:114:29: not found: value sourceId
[error]     name = "stream-reader", sourceId = IdRange(0, nXacts))
[error]                             ^
[error] /home/username0/Chipyard/chipyard/generators/gemmini/src/main/scala/gemmini/DMA.scala:209:20: value Get is not a member of Any
[error]     val get = edge.Get(
[error]                    ^
[error] /home/username0/Chipyard/chipyard/generators/gemmini/src/main/scala/gemmini/DMA.scala:210:7: not found: value fromSource
[error]       fromSource = io.reserve.xactid,
[error]       ^
[error] /home/username0/Chipyard/chipyard/generators/gemmini/src/main/scala/gemmini/DMA.scala:211:7: not found: value toAddress
[error]       toAddress = read_paddr,
[error]       ^
[error] /home/username0/Chipyard/chipyard/generators/gemmini/src/main/scala/gemmini/DMA.scala:212:7: not found: value lgSize
[error]       lgSize = read_lg_size
[error]       ^
[error] /home/username0/Chipyard/chipyard/generators/gemmini/src/main/scala/gemmini/DMA.scala:215:8: value a is not a member of Any
[error]     tl.a.valid := state === s_req_new_block && io.reserve.ready
[error]        ^
[error] /home/username0/Chipyard/chipyard/generators/gemmini/src/main/scala/gemmini/DMA.scala:216:8: value a is not a member of Any
[error]     tl.a.bits := get
[error]        ^
[error] /home/username0/Chipyard/chipyard/generators/gemmini/src/main/scala/gemmini/DMA.scala:218:57: value a is not a member of Any
[error]     io.reserve.valid := state === s_req_new_block && tl.a.ready // TODO decouple "reserve.valid" from "tl.a.ready"
[error]                                                         ^
[error] /home/username0/Chipyard/chipyard/generators/gemmini/src/main/scala/gemmini/DMA.scala:236:14: value a is not a member of Any
[error]     when (tl.a.fire()) {
[error]              ^
[error] /home/username0/Chipyard/chipyard/generators/gemmini/src/main/scala/gemmini/DMA.scala:254:8: value d is not a member of Any
[error]     tl.d.ready := io.beatData.ready
[error]        ^
[error] /home/username0/Chipyard/chipyard/generators/gemmini/src/main/scala/gemmini/DMA.scala:255:29: value d is not a member of Any
[error]     io.beatData.valid := tl.d.valid
[error]                             ^
[error] /home/username0/Chipyard/chipyard/generators/gemmini/src/main/scala/gemmini/DMA.scala:256:35: value d is not a member of Any
[error]     io.beatData.bits.xactid := tl.d.bits.source
[error]                                   ^
[error] /home/username0/Chipyard/chipyard/generators/gemmini/src/main/scala/gemmini/DMA.scala:257:33: value d is not a member of Any
[error]     io.beatData.bits.data := tl.d.bits.data
[error]                                 ^
[error] /home/username0/Chipyard/chipyard/generators/gemmini/src/main/scala/gemmini/DMA.scala:258:39: value d is not a member of Any
[error]     io.beatData.bits.lg_len_req := tl.d.bits.size
[error]                                       ^
[error] /home/username0/Chipyard/chipyard/generators/gemmini/src/main/scala/gemmini/DMA.scala:259:35: value last is not a member of Any
[error]     io.beatData.bits.last := edge.last(tl.d)
[error]                                   ^
[error] /home/username0/Chipyard/chipyard/generators/gemmini/src/main/scala/gemmini/DMA.scala:259:43: value d is not a member of Any
[error]     io.beatData.bits.last := edge.last(tl.d)
[error]                                           ^
[error] /home/username0/Chipyard/chipyard/generators/gemmini/src/main/scala/gemmini/DMA.scala:288:14: not found: value TLHelper
[error]   val node = TLHelper.makeClientNode(
[... 48 lines cut ...]
[error]       ^
[error] /home/username0/Chipyard/chipyard/generators/gemmini/src/main/scala/gemmini/DMA.scala:431:8: value a is not a member of Any
[error]     tl.a.valid := (state === s_writing_new_block || state === s_writing_beats) && !xactBusy.andR()
[error]        ^
[... cut ...]
[error]                                                        ^
[error] /home/username0/Chipyard/chipyard/generators/gemmini/src/main/scala/gemmini/FrontendTLB.scala:42:65: not found: value nWays
[error]   val tlb = Module(new TLB(false, lgMaxSize, TLBConfig(nSets=1, nWays=entries)))
[error]                                                                 ^
[error] 42 errors found
[error] (Compile / compileIncremental) Compilation failed
[error] Total time: 18 s, completed May 7, 2021, 8:12:24 PM
------

How can I import Testchipip?

hngenc, 2021-05-07:
This is unusual. Which command are you using to build Gemmini? Also, have you setup Chipyard? I believe Testchipip is a part of Chipyard.

Z-KN, 2021-05-08:
I use the command "sbt compile" in directory "chipyard/generators/gemmini", but the Testchipip is not in gemmini's default package path.

hngenc, 2021-05-08:
Oh, I see. That's not how you should build Gemmini.

Gemmini needs to be built as part of Chipyard.

You can set up Chipyard by following the instructions here: https://chipyard.readthedocs.io/en/latest/Chipyard-Basics/Initial-Repo-Setup.html#initial-repository-setup

Afterwards, you can build Gemmini by following the instructions here: https://chipyard.readthedocs.io/en/latest/Generators/Gemmini.html

hngenc, 2021-05-08:
Closing the issue for now. Feel free to re-open if you have more questions about the verilog files

closed 2021-05-08 by hngenc