ucb-bar/gemmini #118: Verilator runnning elf failed

AaronJing, 2021-05-08:
Hello!

My branch is dev. When I was loading elf into vertilator simulator (default config but with im2col set true), I meet the following assertion error. I also tried running loop_conv_ws macro in elf, and it failed because of the same error. 
```
[UART] UART0 is here (stdin/stdout).
Output dimension: 9

Randomize inputs...
Randomize weights...
Randomize bias...
CPU conv...
CPU conv took 9994294 cycles
Flatten weights...
Gemmini conv...
[10655] %Error: chipyard.TestHarness.GemminiRocketConfig.top.v:224033: Assertion failed in TOP.TestHarness.chiptop.system.tile_prci_domain.tile_reset_domain.tile.gemmini.spad
%Error: /home/jing/chipyard/sims/verilator/generated-src/chipyard.TestHarness.GemminiRocketConfig/chipyard.TestHarness.GemminiRocketConfig.top.v:224033: Verilog $stop
Aborting...
```

hngenc, 2021-05-08:
Could you check the line number given in the error message to see which assertion was fired?

Also, do you know if the test passes when im2col = false?

AaronJing, 2021-05-08:
An update. After commenting out the assertion, the test failed because of results unmatched.

hngenc, 2021-05-09:
Yeah, the assertion is pretty important; you shouldn't get rid of it.

I can confirm that the `conv` binary works when the simulation is built with VCS. (I assume that's the binary you ran). I can try with Verilator as well, just to make sure.

But first, just to be sure, can you make you that you ran `git submodule update`? Perhaps your `gemmini-rocc-tests` subrepo is not up-to-date? That can cause all kinds of weird error.s

AaronJing, 2021-05-10:
Problem fixed! Conv elf passed verilator. Thanks, I will keep updating `gemmini-rocc-tests` in mind.

hngenc, 2021-05-10:
Great, thanks for the update!

closed 2021-05-10 by hngenc