ucb-bar/gemmini #129: sbt test failed

AaronJing, 2021-06-03:
Hello!
I tried to run the test with the command `sbt test` and received some compilation error. 
```gemmini/src/main/scala/gemmini/AccumulatorMem.scala:93:12: value waddr is not a member of chisel3.Bundle```
I solved this by adding `-Xsource:2.11`. However, It failed compilation again because of the following error.
```
[error] /home/jing/mychipyard/generators/gemmini/src/main/scala/gemmini/DMA.scala:11:8: not found: object testchipip
[error] import testchipip.TLHelper
[error]        ^
[error] /home/jing/mychipyard/generators/gemmini/src/main/scala/gemmini/DMA.scala:116:14: not found: value TLHelper
[error]   val node = TLHelper.makeClientNode(
[error]              ^
[error] /home/jing/mychipyard/generators/gemmini/src/main/scala/gemmini/DMA.scala:117:5: value name_= is not a member of gemmini.StreamReaderCore[T,U,V]
[error]     name = "stream-reader", sourceId = IdRange(0, nXacts))
[error]     ^
[error] /home/jing/mychipyard/generators/gemmini/src/main/scala/gemmini/DMA.scala:117:29: not found: value sourceId
[error]     name = "stream-reader", sourceId = IdRange(0, nXacts))
[error]                             ^
[error] /home/jing/mychipyard/generators/gemmini/src/main/scala/gemmini/DMA.scala:190:20: value Get is not a member of Any
[error]     val get = edge.Get(
[error]                    ^
```
Could you give some hints to solve this, please? Thanks.

hngenc, 2021-06-03:
We don't use `sbt test`. We usually use the tests in `gemmini-rocc-tests` instead.

What tests were you trying to run exactly?

AaronJing, 2021-06-04:
All good. I just found out the tests were made 2 years ago. Thanks though!

closed 2021-06-04 by AaronJing

hngenc, 2021-06-04:
No problem!