ucb-bar/gemmini #137: Scaling doesn't work when loading i32 to the accumulator in spike

yamaguchi1024, 2021-07-02:
The scale value is being defaulted to 1.0 no matter what mvin scale was passed to `gemmini_extended3_config_ld`

Simple reproducer:
https://gist.github.com/yamaguchi1024/8eadb86f792bdba55087e16003df70d2
The code is loading 16x16 i32 1s to accumulator with scale=4.0, and storing i32 data from accumulator to DRAM.
Compile ^ these files with riscv64-unknown-elf-gcc, run spike and you would get

```
spike --extension=gemmini gemmini_build/ldst_acc_i32_16                          
Gemmini extension configured with:
    dim = 16                    
Results Don't Match             
Correct Result (res):           
4 4 4 4 4 4 4 4 4 4 4 4 4 4 4 4 
4 4 4 4 4 4 4 4 4 4 4 4 4 4 4 4 
4 4 4 4 4 4 4 4 4 4 4 4 4 4 4 4 
4 4 4 4 4 4 4 4 4 4 4 4 4 4 4 4                                                                                                                  
4 4 4 4 4 4 4 4 4 4 4 4 4 4 4 4                                                                                                                  
4 4 4 4 4 4 4 4 4 4 4 4 4 4 4 4                                                                                                                  
4 4 4 4 4 4 4 4 4 4 4 4 4 4 4 4                                                                                                                  
4 4 4 4 4 4 4 4 4 4 4 4 4 4 4 4                                                                                                                  
4 4 4 4 4 4 4 4 4 4 4 4 4 4 4 4                                                                                                                  
4 4 4 4 4 4 4 4 4 4 4 4 4 4 4 4                                                                                                                  
4 4 4 4 4 4 4 4 4 4 4 4 4 4 4 4                                                                                                                  
4 4 4 4 4 4 4 4 4 4 4 4 4 4 4 4                                                                                                                  
4 4 4 4 4 4 4 4 4 4 4 4 4 4 4 4                                                                                                                  
4 4 4 4 4 4 4 4 4 4 4 4 4 4 4 4                                                                                                                                             
4 4 4 4 4 4 4 4 4 4 4 4 4 4 4 4                                                                                                                                             
4 4 4 4 4 4 4 4 4 4 4 4 4 4 4 4                                                                                                                                             
Computed Roundtrip (y):                                                                                                                                                     
1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1                                                                                                                                             
1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1                                                                                                                                             
1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1                                                       
1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1                                                                                                                                             
1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1                                                                                                                                             
1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1                                                                                                                                             
1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1                                                       
1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1                                                                                                                                             
1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1                                                       
1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1                                                                                                                                             
1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1            
1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1            
1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1            
1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1            
1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1            
1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1            
*** FAILED *** (tohost = 1)  
```

hngenc, 2021-07-09:
It seems this might actually be an issue with the default `gemmini_params.h`. Can you try copying this `gemmini_params.h` file to both `gemmini-rocc-tests/include` and `toolchains/esp-tools/riscv-isa-sim/gemmini/gemmini_params.h`,  and then running your test again?

(The file has a `.h.txt` extension due to GitHub's file-upload requirements, but you should just remove the `.txt` from the filename).

[gemmini_params.h.txt](https://github.com/ucb-bar/gemmini/files/6788976/gemmini_params.h.txt)

yamaguchi1024, 2021-07-14:
@hngenc Sorry for the late response.
I tried your gemmini_params.h, it fixed the scaling error when loading i32 to the accumulator in our unittest. Thank you so much!
Let me try to run the whole test with this change and confirm that it works.

closed 2021-11-11 by hngenc