ucb-bar/gemmini #140: process to generate C headers and run ISA simulation?

fw-tamu, 2021-07-07:
Hi:
I'm interested to modify gemmini's parameters and  run ISA simulation using Spike. What's the correct procedure for that?
The manuals says "The Gemmini generator generates a C header file based on the generator parameters" but it doesn't specify which commands.
If i understand correctly, the chipyard manual is focused on RTL generation of a whole SoC. Do I need to form a dummy SoC shell that only contains a gemmini accelerator and use the "make" command under chipyard/sims/verilator, in order to generate the C header files (mainly gemmini_params.h, which i'll need to compile rocc-test applications)? 
Thanks

hngenc, 2021-07-08:
The documentation for running Spike simulations can be found here: https://chipyard.readthedocs.io/en/latest/Generators/Gemmini.html#build-and-run-gemmini-tests

However, before running through those steps, check `SPIKE.hash` to see which Spike commit you need to be on. Afterwards, go to `toolchains/esp-tools/riscv-isa-sim/build`, checkout the correct commit, and run `make && make install` to build the correct Spike version.

fw-tamu, 2021-07-08:
> The documentation for running Spike simulations can be found here: https://chipyard.readthedocs.io/en/latest/Generators/Gemmini.html#build-and-run-gemmini-tests
>
> However, before running through [...]

How do i re-generate the "gemmini_params.h" after changing parameters in "gemmini/src/main/scala/gemmini/Configs.scala"?

Thanks

hngenc, 2021-07-09:
You need to build the cycle-accurate simulator to do that, unfortunately (either Verilator or VCS). The good news is that you don't need to wait for the full build to finish. The build process generates a new `gemmini_params.h` very early, and you can kill the build process after that if you're in a hurry.

fw-tamu, 2021-07-09:
Hi:
After i changed the parameters in Configs.scala,
I executed "make" under ~/chipyard/sims/verilator;
yet i still don't see the "gemmini_params.h"  even after the "make" completes.

hngenc, 2021-07-09:
It will update the `gemmini_params.h` file in `gemmini-rocc-tests/include`, rather than creating a new one in the local directory

fw-tamu, 2021-07-09:
> It will update the `gemmini_params.h` file in `gemmini-rocc-tests/include`, rather than creating a new one in the local directory

that's where i'm looking at, but it was not updated (same old modification date)

hngenc, 2021-07-09:
I see. What was the full `make` command you used in `sims/verilator`?

fw-tamu, 2021-07-09:
> I see. What was the full `make` command you used in `sims/verilator`?

I simply typed "make", should i add any argument/flags?

hngenc, 2021-07-09:
Try this command: `make CONFIG=GemminiRocketConfig`

Afterwards, make sure to copy your new `gemmini_params.h` file into `toolchains/esp-tools/riscv-isa-sim/gemmini/gemmini_params.h` and then run `make && make install` in  ``toolchains/esp-tools/riscv-isa-sim/build`

fw-tamu, 2021-07-09:
> Try this command: `make CONFIG=GemminiRocketConfig`
> 
> Afterwards, make sure to copy your new `gemmini_params.h` file into `toolchains/esp-tools/riscv-isa-sim/gemmini/gemmini_params.h` and then run `make && make install` in `toolchains/esp-tools/riscv-isa-sim/build`

I used your command but still no luck, it ran into an error and failed. The log is attached:
[make.log](https://github.com/ucb-bar/gemmini/files/6789054/make.log)

hngenc, 2021-07-09:
This is the relevant line in the log:

> [error] Caused by: java.lang.IllegalArgumentException: requirement failed: the number of rows in an accumulator bank must be a multiple of the dimensions of the systolic array

Sorry it's buried so deep in the log.

Can you update your config to meet that requirement?

fw-tamu, 2021-07-09:
> This is the relevant line in the log:
>
>
> Sorry it's buried so deep in the log.
>
> Can you update your config to meet that requirement?

Yes it works after i fix that error. Thanks

hngenc, 2021-07-09:
No problem! Let us know if you have any more problems.

closed 2021-07-09 by hngenc