ucb-bar/gemmini #143: Generalizability of Gemmini generator

johnny-wang16, 2021-08-30:
Thank you for your amazing work! I read the paper and have some questions regarding the generalizability of Gemmini generator. Say today I'd like to change the design on the schematic in the README page, for example:  replace the L2 cache  with an SRAM and remove the DRAM below completely or just have 1 tile of 32x32 PE array, could Gemmini generator accommodate changes like this? Thanks.

hngenc, 2021-09-11:
Sorry for the late response!

> replace the L2 cache with an SRAM  and remove the DRAM below completely

I believe the L2 is already implemented as an SRAM.  Do you mean that you want to remove the L2 and DRAM and replace them with just the Gemmini scratchpad?

It might be possible to do this, but we haven't tried to build such a self-contained system before. Our memory system is based off of [RocketChip](https://github.com/chipsalliance/rocket-chip); you could try asking them if it's possible to remove the L2 and DRAM. We haven't tried doing that ourselves, and it's not one of our default configuration options.

> just have 1 tile of 32x32 PE array

Yes, this change is supported. Just change `mesh_rows` and `mesh_columns` in the Config options.

johnny-wang16, 2021-09-12:
Got it. Thanks for the reply!

johnny-wang16, 2021-09-13:
Hi Hasan, should I also change the DIM size in gemmini_params.h as well if I just want to have 1 tile of 32x32 PE array? Thanks!

hngenc, 2021-09-13:
Nope; if you compile the Verilator, VCS, or Firesim model after changing the configs, then the `DIM` size in `gemmini_params.h` should automatically be updated for you.

However, you _will_ need to update Spike's `gemmini_params.h` manually, if you're planning to run on Spike.

Running this would be sufficient, **after** `gemmini_params.h` has been updated (assuming that `~/CHIPYARD_DIR` is your path to Chipyard):

```
cd $CHIPYARD_DIR/toolchains/esp-tools/riscv-isa-sim/
cp $CHIPYARD_DIR/generators/gemmini/software/gemmini-rocc-tests/include/gemmini_params.h gemmini/gemmini_params.h
cd build && make && make install
```

johnny-wang16, 2021-09-13:
Got it! Thank you so much for the response!

swc1326, 2021-09-16:
Hi, @hngenc  I also want to modify the configuration of the Gemmini and face some similar issues.
> Yes, this change is supported. Just change `mesh_rows` and `mesh_columns` in the Config options.

So I've changed those parameters in `Config.scala` as follow,
```
tileRows = 4,
tileColumns = 4,
meshRows = 4,
meshColumns = 4,
```
But it does not work when I try `make debug CONFIG=GemminiRocketConfig` under the path `/chipyard/ sims/ verilator`. Do I have to change any other parameters as well?

hngenc, 2021-09-17:
@swc1326 That seems to be an issue with [Chipyard](https://github.com/ucb-bar/chipyard), rather than with Gemmini. Gemmini provides the RTL, but Chipyard builds the actual cycle-accurate simulators based on that RTL.

Can you try opening an issue on Chipyard's GitHub page and dropping a link to that issue here?

johnny-wang16, 2021-11-05:
Turned out the above issue was due to insufficient memory size in Oracle virtual box. Increasing the size of RAM for vbox solved it.

johnny-wang16, 2021-11-05:
hi @hngenc , the image below illustrates what i meant by replacing L2 cache. Basically, I'd like to take out the L2 cache  completely due to area constraints and just have rocket core,  DRAM and gemmini on my SoC. Currently, Im able to compile a version with the configuration i just mentioned. But Im blocked immediately after entering the function `tiled_conv_A_stride_auto` . I know this is probably too little information for you to debug but could you provide some high level suggestions as to what I configurations i need to change/documents I should go over ? Thank you very much!

> Sorry for the late response!
>
>
> I believe the L2 is already implemented as an SRAM. Do you mean that you want to remove the L2 and DRAM and replace them with just the Gemmini scratchpad?
>
> It might be [...]

[image]

closed 2021-11-12 by johnny-wang16