ucb-bar/gemmini #151: Im2col Function Problem

swc1326, 2021-10-12:
Hi, I want to use the image to column function, so I've change the Config.scala as below,
```
    hasIm2col = true, //declare im2col block

    sp_banks = 2,
    sp_singleported = false,
    acc_banks = 1,
    acc_singleported = true,
    num_acc_sub_banks = 4,
    sp_capacity = CapacityInKilobytes(32),
    shifter_banks = 1, // TODO add separate parameters for left and up shifter banks
    dataflow = Dataflow.WS,
    acc_capacity = CapacityInKilobytes(16),
    mem_pipeline = 4,
    dma_maxbytes = 64, // TODO get this from cacheblockbytes
    dma_buswidth = 128, // TODO get this from SystemBusKey
    aligned_to = 1,
    tlb_size = 4,
    use_tlb_register_filter = true,
    max_in_flight_reqs = 16,
    use_dedicated_tl_port = false,
```
but when I run `./simulator-chipyard-GemminiRocketConfig-debug`, it comes out an error
```
[1223000] %Error: chipyard.TestHarness.GemminiRocketConfig.top.v:285131: Assertion failed in TOP.TestHarness.chiptop.system.tn.tile_reset_domain.tile.gemmini.rob
%Error: /chipyard/sims/verilator/generated-src/chipyard.TestHarness.GemminiRocketConfig/chipyard.TestHarness.Gemmini.top.v:285131: Verilog $stop
Aborting...
```
Which part should I modify?

mbelda, 2022-12-15:
Hello,

Is the im2col unit already fixed? 
I am trying to obtain the mentioned speedup of more than x1000 with the im2col performed on Gemmini as in the dac paper for a tunned cnn.

Also, I am running the inference using onnx. Is the im2col execution on Gemmini enabled by default? Or do I have to use any  onnx special configuration option or something?

Thanks!

hngenc, 2022-12-22:
We basically just do im2col on Gemmini by default now. Everything that the old im2col unit was doing has been subsumed by the [LoopConv](https://github.com/ucb-bar/gemmini/blob/master/src/main/scala/gemmini/LoopConv.scala#L1157) and [PixelRepeater](https://github.com/ucb-bar/gemmini/blob/master/src/main/scala/gemmini/PixelRepeater.scala#L28) modules.

If you want to run im2col on the CPU, and not use Gemmini's im2col features, then you can run some of the binaries in `gemmini-rocc-tests` with the `matmul` command-line argument, as seen [here](https://github.com/ucb-bar/gemmini-rocc-tests/blob/dev/imagenet/resnet50.c#L49). That will perform im2col on the CPU, and then run matmuls on Gemmini.

I don't think ONNX-Runtime supports this option by default though.