ucb-bar/gemmini #159: Assertion failed: 'A' channel carries PutPartial type which is unexpected using diplomatic parameters (connected to CrossingHelpler.scala:61:80)

johnny-wang16, 2021-11-12:
[image]
Hi, I was getting this error when running the provided baremetal tests like `matmul `and `conv` in verilator. the main configuration I changed is as follows : 1) The default template contains a 512k L2 cache and a DRAM. I removed the L2 cache completely and set the DRAM size to 128k. 2) I changed the[ default Rocket Chip system](https://chipyard.readthedocs.io/en/latest/Generators/Rocket-Chip.html) protocol from AXI to AHB (so I have TL to AHB -> AHB Mem and TL to AHB -> AHB slave). I was able to pinpoint that this error appears during the move out process after Gemmini finishes matrix multiplication. I've checked the output (stored in the accumulator) of the matrix multiplication that Gemmini calculates is correct. The DRAM address that `mvout ` intends to carry the accumulator output to is also a valid address. So I'm confused why (and where) this assertion failed happens. Is there any documentation that explains the reason behind this kind of assertions? Thanks!

hngenc, 2021-11-12:
That's a TileLink assertion that you're seeing, rather than a Gemmini assertion. I'm guessing the most likely reason is because you removed the L2. I don't believe we've ever tried doing that ourselves.

hngenc, 2021-11-12:
I take back what I said earlier. I just created my own config without an L2, and it seems to work just fine. It passes the baremetal tests, at least on VCS. (I haven't tried running with Verilator yet, but I doubt the result would be different than VCS).

@johnny-wang16 Perhaps the error is caused by your use of AHB? That is also something that we have not tried doing before.

jerryz123, 2021-11-12:
It would be useful to know which Tilelink channel is causing the TLMonitor assertion. If you report the path to the assertion (should be visible in .out and .log files) it would be helpful.

But the most likely cause is the AHB setup you have. Does the AHB system work when there is no Gemmini, and just the core?

johnny-wang16, 2021-11-12:
Hi all, thanks for your prompt response. I'd like to update the situation. So, from the [TileLink Specification](https://sifive.cdn.prismic.io/sifive%2Fcab05224-2df1-4af8-adee-8d9cba3378cd_tilelink-spec-1.8.0.pdf) page 36, opcode 1 should be a valid opcode so I'm not sure what the assertion is complaining about. (maybe sth about diplomatic parameters ? but what is a diplomatic parameter?). I also tried a setting that only removed L2 but using default AXI bus and it works on verilator. So I think removing L2 is not the cause and the assertion error should be sth to do with AHB bus. On the configuration that has L2 removed and with AHB bus, I tried commenting out the assertions in the TileLink Monitor.scala file and I can see from the waveforms that it tries to write the correct value to the correct DRAM address but the write operation takes 2 cycles and maybe overwriting the correct value. Anyway I think removing L2 is safe and the assertion is caused by AXI to AHB.

johnny-wang16, 2021-11-12:
> It would be useful to know which Tilelink channel is causing the TLMonitor assertion. If you report the path to the assertion (should be visible in .out and .log files) it would be helpful.
>
> But the [...]

Hi @jerryz123  Thanks for the advice. The Tilelink channel that's causing the assertion is channel A and the log is the image I pasted when raising this issue. As to whether my AHB system works when there's no Gemmini, I haven't tried it. But I wrote a baremetal test that re-arranged the input image (slice 1 square image to 4 equal squares) before passing it to neural network . This test only involves CPU and DRAM and since that path works, I think the communication between CPU and DRAM using AHB bus should be fine. I'm suspecting that there's some chipyard configuration regarding the TLMonitor that needs to be changed when switching to AHB. ill look into this. But thank you guys for your valuable advice on this issue.

johnny-wang16, 2021-11-24:
https://github.com/chipsalliance/rocket-chip/issues/2909#issue-1062087065

closed 2021-11-24 by johnny-wang16