ucb-bar/gemmini #161: [Question] Possibility of allowing non-square input/kernel for convolution

johnny-wang16, 2021-11-15:
The `LOOP_CONV ` module assumes input image and convolution kernels are square matrices. This restriction might not hold all the time (ex: 1. input image might not always be square. 2. for audio signals, convolution needs to be in 1D). I was wondering if there are any plans to expand the capability of Gemmini to allow non-square inputs (or even kernels). 

I asked this because matmul using `LOOP_MATMUL_WS ` module seems to be a lot faster and have much less cycle count comparing to it's output stationary counterparts that's not implemented as a hardware module. Although it's possible to implement 1D-convolution function in C following the load -> compute -> store scheme (like how `tiled_matmul_os` is implemented) , I'm worrying about the CPU overheads this might bring. As I'm still pretty new to Chisel and wasn't able to develop a hardware module for 1D convolution myself in the near future, I'm wondering if there are any plans for supporting non-square inputs or kernels. (Also, are there any documentations for the scala files like LoopConv or LoopMatmul.) Thanks!

hngenc, 2021-11-17:
I think we should be able to support this without too much trouble. @simonguozirui will take a stab at making a PR for this next week.

johnny-wang16, 2021-11-18:
That'd be awesome! Thanks !