ucb-bar/gemmini #163: [Question] inconsistency between gemmini.h and readme on config_mvin and config_mvout ISA

johnny-wang16, 2021-11-16:
At the [config_mvin configures the Load pipeline](https://github.com/ucb-bar/gemmini/tree/v0.5#config_mvin-configures-the-load-pipeline) section of readme, it says  " `rs1[0:1]` must be 01 " which means 0th bit is 0 and 1st bit is 1, which means a 2 in binary representation. However, in `gemmini.h` , [line 54](https://github.com/ucb-bar/gemmini-rocc-tests/blob/3aaa230733a9eba6edf4d14243d84595e017522f/include/gemmini.h#L54) , `CONFIG_LD ` is defined to be 1. Same thing happens with [config_mvout configures the Store pipeline](https://github.com/ucb-bar/gemmini/tree/v0.5#config_mvout-configures-the-store-pipeline) and [line 55](https://github.com/ucb-bar/gemmini-rocc-tests/blob/3aaa230733a9eba6edf4d14243d84595e017522f/include/gemmini.h#L55) in `gemmini.h` . Is this a typo in readme ? Thanks in advance!

hngenc, 2021-11-17:
Good catch! That's a typo. We should define it as `rs1[1:0]` instead.

I'll make a PR to fix that

hngenc, 2021-11-18:
Let's keep this open till I merge a fix for it :)

hngenc, 2021-11-21:
Fixed by `6c620f5` on `master`

closed 2021-11-21 by hngenc