ucb-bar/gemmini #166: 100% TLB hit rate collected from gemmini counter

yuqin-jz, 2021-11-28:
I tried to get the TLB miss rate when running MLP test and conv.c test using verilator. But all of these hit rate are 100%. What I did in test file is to add  
counter_configure(0, DMA_TLB_MISS_CYCLE);
counter_configure(1, DMA_TLB_HIT_REQ);
counter_configure(2, DMA_TLB_TOTAL_REQ);
in the beginning of the test and use counter_read(0,value) to get the data. But the DMA_TLB_MISS_CYCLE is always zero in mlp and conv.v test. I attached the test file and the result here.

I also tried to increase the size of input so that it is bigger than TLB size and scratchpad memory size. But it seems there is still no miss, which is not right. I am not sure whether TLB is bypassed or disabled. And I don't understand why the hit rate is always 100%. Thank you so much!

[mlp2_result.txt](https://github.com/ucb-bar/gemmini/files/7612630/mlp2_result.txt)
[mlp2.txt](https://github.com/ucb-bar/gemmini/files/7612631/mlp2.txt)

shubhamn94, 2021-11-28:
Even I tried on my system too by pulling the docker image (and corresponding pre-built binary) provided in the tutorial slides. https://sites.google.com/berkeley.edu/gemminitutorialiiswc2021/
I am also getting same error. TLB miss cycles are 0. Seems like 100% hit. 
I modified the gemmini_counter.c to configure and print DMA_TLB_MISS_CYCLE register and other few registers but multiple of those are zero. I also made sure DISABLE counter is set to 0. i.e. none of the counter is disabled. I also tried **Flushing** the TLB using **gemmini_flush(0)**, **gemmini_flush(1)** and randomly **putting fences** in between as the note in the github repo mentions that flush happens immediately as soon as it is encountered. So try putting fences in-between.
Please find the modified gemmini_counter.c attached.
Looking forward for the solution of this. Thanks! 

**Output**
```
root@ee312b25d4dc:~/chipyard/generators/gemmini# ./scripts/run-verilator.sh gemmini_counter
../../generators/gemmini/software/gemmini-rocc-tests/build/bareMetalC/gemmini_counter-baremetal
This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.
Listening on port 42373
[UART] UART0 is here (stdin/stdout).
DMA TLB MISS cycles: 0
DMA TLB HIT REQ: 0
DMA TLB TOTAL REQ: 0
RDMA_TLB_WAIT_CYCLES: 0
WDMA_TLB_WAIT_CYCLES: 0
RDMA_ACTIVE_CYCLE: 0
WDMA ACTIVE CYCLE: 0
Read DMA cycles: 188
DMA TLB MISS cycles: 0
DMA TLB HIT REQ: 18
DMA TLB TOTAL REQ: 18
RDMA_TLB_WAIT_CYCLES: 0
WDMA_TLB_WAIT_CYCLES: 0
RDMA_ACTIVE_CYCLE: 230
WDMA ACTIVE CYCLE: 198
Cycle when taking snapshot: 198, Cycle read after operation finished: 198
Cycles after snapshot is reset: 303
Cycles after counter reset: 0
DMA TLB MISS cycles: 0
DMA TLB HIT REQ: 6
DMA TLB TOTAL REQ: 6
RDMA_TLB_WAIT_CYCLES: 0
WDMA_TLB_WAIT_CYCLES: 0
RDMA_ACTIVE_CYCLE: 180
RESERVATION_STATION # of load insts after executing 7 mvin and mvout insts: 36
```

```
//flush
#define gemmini_flush(skip) \
  ROCC_INSTRUCTION_RS1_RS2(XCUSTOM_ACC, skip, 0, k_FLUSH)
  
 #define ROCC_INSTRUCTION_RS1_RS2(x, rs1, rs2, funct) \
 ROCC_INSTRUCTION_0_R_R(x, rs1, rs2, funct)
```

[image]
[gemmini_counter.txt](https://github.com/ucb-bar/gemmini/files/7612727/gemmini_counter.txt)

hngenc, 2021-12-05:
I've been able to replicate this issue; and I'm trying to debug it. Hopefully, I'll have a fix merged soon.

hngenc, 2021-12-05:
The main issue here seems to be that in baremetal mode on VCS/Verilator, virtual memory is not enabled. Physical addresses and virtual addresses are exactly the same.

Therefore, the TLB is really just doing a no-op and returning the same address that was given to it. There's no table lookup happening, no page table walks, and no TLB "misses".

Firesim simulations, which actually run Linux and a virtual memory system, should not have this problem. Firesim simulations, however, may have an unrelated error counting TLB hits, which I'm fixing now in #168. Hopefully, #168 will be merged into `dev` within the next day or two.

After #168 is merged, you should be able to count TLB hits/misses accurately in Firesim simulations.

hngenc, 2021-12-06:
@yuqin-jz @shubhamn94 I've merged #168 into the `dev` branch. If you run your tests on Firesim now, you should get more reasonable TLB hit/miss numbers.

I'm going to close this issue for now, but please feel free to re-open it if you run into any issues getting reasonable TLB hit/miss numbers on Firesim.

hngenc, 2021-12-13:
@yuqin-jz @shubhamn94

I've just added new scripts to `dev` which should resolve your issue.

Try running these commands:

```
git checkout dev && git pull && git submodule update

cd software/gemmini-rocc-tests
./build.sh

cd -
./scripts/build-verilator.sh
./scripts/run-verilator.sh --pk $YOUR_BINARY
```

Where `$YOUR_BINARY` is the binary you built.

I've also added [an explanation here](https://github.com/ucb-bar/gemmini/tree/dev#building-gemmini-software) of the different kinds of Gemmini programs you can compile, and whether or not they support virtual memory.

Let me know if this all works for you! I'm planning to merge this in to `master` soon. I'll close this issue when it's been merged in.

And thanks for bringing this to my attention!

hngenc, 2021-12-14:
I just merged these changes into `master`, so I'm closing this issue again.

Let me know if you still get weird TLB results after these recent changes. You should be able to run on Verilator with virtual memory enabled now.

PR #168 (2021-12-06): Fix tlb hit counter
Fixes bug where TLB hits were being counted incorrectly.

Prior to this PR, we were using `RegNext(io.req.fire())` to match TLB requests to TLB responses. However, we made our interface to the TLB combinational months ago, so the `RegNext` is no longer necessary (and is actually incorrect).

Closes #166
closed 2021-12-14 by hngenc