ucb-bar/gemmini #180: /root/chipyard/common.mk:137: recipe for target 'firrtl_temp' failed

6357o31337, 2021-12-28:
Hello.

When I 'Building Gemmini Functional Simulators'
(
cd chipyard/generators/gemmini
./scripts/build-verilator.sh
)

/root/chipyard/common.mk:137: recipe for target 'firrtl_temp' failed
make: *** [firrtl_temp] Killed

[image]

How can I solve?

Thank you.

hngenc, 2021-12-28:
Can you send me the output of these commands?

```
cd ~/chipyard
git log -n1

cd ~/chipyard/generators/gemmini
git log -n1
```

hngenc, 2022-01-19:
Closing due to lack of updates. Feel free to re-open if you run into issues building the Verilator simulator again

eneneli, 2022-01-19:
Hi, I'm having the same problem. I've answered your question below, hope that helps.

> Can you send me the output of these commands?
> 
> cd ~/chipyard
> git log -n1

commit ec1b075658fb92a624151536dd1de76bad94f51f (HEAD)
Merge: e6f76235 54447d03
Author: David Biancolin <david.biancolin@gmail.com>
Date:   Wed Dec 8 14:04:25 2021 -0800

    Merge pull request #1054 from ucb-bar/use-published-deps
    
    Use Published Chisel, FIRRTL, Treadle, FIRRTLInterpreter packages instead of building from source.

> cd ~/chipyard/generators/gemmini
> git log -n1

commit 72461235cc955a48f45ae3fefc7dfd68e27bc284 (HEAD, tag: v0.6.2, origin/master, origin/HEAD, master)
Author: Hasan Genc <hngenc@berkeley.edu>
Date:   Mon Dec 20 03:19:43 2021 -0800

    Update README to checkout v0.6.2

hngenc, 2022-01-19:
Thanks, @eneneli. Those commits _should_ work. There may be some issue with your environment?

Could you send me the full output of your `build-verilator.sh` command? That might help me to figure this out.

eneneli, 2022-01-20:
Thanks for taking a look. Here's the output:

```
~/chipyard/generators/gemmini$ ./scripts/build-verilator.sh 
Running with RISCV=/home/eneli/chipyard/esp-tools-install
cd /home/eneli/chipyard && java -jar /home/eneli/chipyard/generators/rocket-chip/sbt-launch.jar   ";project tapeout; runMain barstools.tapeout.transforms.GenerateTopAndHarness  --output-file /home/eneli/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.top.v --harness-o /home/eneli/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.harness.v --input-file /home/eneli/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.fir --syn-top ChipTop --harness-top TestHarness --annotation-file /home/eneli/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.anno.json --top-anno-out /home/eneli/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.top.anno.json --top-dotf-out /home/eneli/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/firrtl_black_box_resource_files.top.f --top-fir /home/eneli/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.top.fir --harness-anno-out /home/eneli/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.harness.anno.json --harness-dotf-out /home/eneli/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/firrtl_black_box_resource_files.harness.f --harness-fir /home/eneli/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.harness.fir --infer-rw --repl-seq-mem -c:TestHarness:-o:/home/eneli/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.top.mems.conf -thconf /home/eneli/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.harness.mems.conf --target-dir /home/eneli/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig --log-level error "
Picked up JAVA_TOOL_OPTIONS: -Xmx8G -Xss8M -XX:MaxPermSize=256M -Djava.io.tmpdir=/home/eneli/chipyard/.java_tmp
OpenJDK 64-Bit Server VM warning: Ignoring option MaxPermSize; support was removed in 8.0
[info] welcome to sbt 1.4.9 (Ubuntu Java 11.0.13)
[info] loading settings for project chipyard-build from plugins.sbt ...
[info] loading project definition from /home/eneli/chipyard/project
[info] loading settings for project chipyardRoot from build.sbt ...
[info] loading settings for project hardfloat from build.sbt ...
[info] loading settings for project rocketConfig from build.sbt ...
[info] loading settings for project chisel_testers from build.sbt ...
[info] loading settings for project testchipip from build.sbt ...
[info] loading settings for project icenet from build.sbt ...
[info] loading settings for project hwacha from build.sbt ...
[info] loading settings for project boom from build.sbt ...
[info] loading settings for project cva6 from build.sbt ...
[info] loading settings for project ibex from build.sbt ...
[info] loading settings for project sodor from build.sbt ...
[info] loading settings for project gemmini from build.sbt ...
[info] loading settings for project mdf from build.sbt ...
[info] loading settings for project barstoolsMacros from build.sbt ...
[info] loading settings for project sim-build from plugins.sbt ...
[info] loading project definition from /home/eneli/chipyard/sims/firesim/sim/project
[info] loading settings for project firesim from build.sbt ...
[info] loading settings for project targetutils from build.sbt ...
[info] loading settings for project midas from build.sbt ...
[info] resolving key references (43040 settings) ...
[info] set current project to chipyardRoot (in build file:/home/eneli/chipyard/
[info] set current project to tapeout (in build file:/home/eneli/chipyard/)
[info] running barstools.tapeout.transforms.GenerateTopAndHarness --output-file /home/eneli/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.top.v --harness-o /home/eneli/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.harness.v --input-file /home/eneli/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.fir --syn-top ChipTop --harness-top TestHarness --annotation-file /home/eneli/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.anno.json --top-anno-out /home/eneli/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.top.anno.json --top-dotf-out /home/eneli/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/firrtl_black_box_resource_files.top.f --top-fir /home/eneli/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestH [... cut ...] /chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.top.mems.conf -thconf /home/eneli/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.harness.mems.conf --target-dir /home/eneli/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig --log-level error
make: *** [/home/eneli/chipyard/common.mk:137: firrtl_temp] Killed
```

hngenc, 2022-01-20:
The `Killed` in your error message suggests to me that you simply ran out of RAM.

Chipyard will try to take up to 8G of RAM (which you can see in `JAVA_TOOL_OPTIONS` in your error message). Perhaps your computer has less than 8G of RAM?

Some users ran into a similar issue here: https://github.com/ucb-bar/chipyard/issues/455

I'm not sure if there is any solution besides finding a more powerful server to work on.

eneneli, 2022-01-20:
I switched to a machine with more RAM and got further this time, but now the Verilator build gets stuck in a loop with the following warning:

```
 <command-line>: note: this is the location of the previous definition
g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -O3 -std=c++11 -I/home/eneli/chipyard/esp-tools-install/include -I/home/eneli/chipyard/tools/DRAMSim2 -I/home/eneli/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig    -D__STDC_FORMAT_MACROS -DTEST_HARNESS=VTestHarness -DVERILATOR -include /home/eneli/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.plusArgs -include /home/eneli/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/verilator.h -include /home/eneli/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig/VTestHarness.h  -DVL_THREADED -std=gnu++14  -c -o VTestHarness__119.o VTestHarness__119.cpp
In file included from /usr/local/share/verilator/include/verilated.h:27,
                 from /usr/local/share/verilator/include/verilated_vcd_c.h:23,
                 from /home/eneli/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/verilator.h:4,
                 from <command-line>:
/usr/local/share/verilator/include/verilatedos.h:252: warning: "__STDC_FORMAT_MACROS" redefined
  252 | #define __STDC_FORMAT_MACROS
      | 

```

hngenc, 2022-01-20:
How long was it in this loop? Verilator can be pretty slow to compile sometimes.

eneneli, 2022-01-20:
Ah, I see. Yes, I left it running and it finished eventually. I could run the template too without issues. So it's all working now, thank you for your help!

hngenc, 2022-01-20:
Great! VCS is usually faster to compile, if you can get your hands on a license.

Closing the issue for now, but feel free to re-open if you have issues again with building the simulator.

closed 2022-01-20 by hngenc