ucb-bar/gemmini #181: matrix_add only works for matrices with the same "DIM"

johnny-wang16, 2022-01-03:
Ideally, a matrix-adding function should work for matrices with arbitrary dimensions (or at least square matrices with arbitrary dimensions). The current method for adding matrices in  `matrix_add.c` only works when input matrices has the same dimension as the PE elements (indicated by the variable `DIM`). I'm wondering if there's an easy way to add matrices with dimensions greater/smaller than the dimension of PE elements.

hngenc, 2022-01-03:
Would the `tiled_resadd_auto` function work for you? You can see an example of it in `resadd.c`. It should be able to add two arbitrarily-sized matrices.

johnny-wang16, 2022-01-04:
thanks for the info! I'll try it out.

hngenc, 2022-01-18:
Closing this for now, but feel free to re-open if you have any questions about `resadd`

closed 2022-01-18 by hngenc