ucb-bar/gemmini #182: How to efficiently scale a matrix?

johnny-wang16, 2022-01-04:
I'd like to uniformly scale a matrix with arbitrary dimension by a constant value. My current implementation uses tiled_matmul and multiplies the  matrix A with an identity matrix B and I set the scale factor to be the constant value I want. Is there a more efficient implementation of achieving this functionality ? Theoretically, it can be achieved using gemmini by mvin/ mvout scale right ?

hngenc, 2022-01-04:
`mvin_scale` should be what you're looking for. In [this line](https://github.com/ucb-bar/gemmini-rocc-tests/blob/dev/bareMetalC/mvin_scale.c#L87) we set the scale factor to be `(n+1)`, which is a scalar value.

The scaling is done during the mvin, and then you can just mvout the correct result. No need to do a matrix multiplication.

hngenc, 2022-01-18:
Closing this for now, but feel free to re-open if you have any questions about `mvin_scale`

closed 2022-01-18 by hngenc