ucb-bar/gemmini #183: System freeze on FPGA VCU118

L-Chocoland, 2022-01-11:
I try to build Gemmini on VCU118 followed Chipyard guide.
When i try the default config of Gemmini, and run linux, it works well on fpga.
But when i change some parameter(e.g. 32*32 PE array),system will freeze when running test.
it seems like https://github.com/ucb-bar/gemmini/issues/154 ,but the default config doesn't have problem.

for example:
running template.c in 32*32 PE Array:

Flush Gemmini TLB of stale virtual addresses
Initialize our input and output matrices in main memory
Calculate the scratchpad addresses of all our matrices
  Note: The scratchpad is "row-addressed", where each address contains one matrix row
Move "In" matrix from main memory into Gemmini's scratchpad
Move "Identity" matrix from main memory into Gemmini's scratchpad
Multiply "In" matrix with "Identity" matrix with a bias of 0
Move "Out" matrix from Gemmini's scratchpad into main memory

then system freeze,fpga should be restart.
it seems mv_out have some problem,when i changing parameters.

How can i fix it?

hngenc, 2022-01-11:
Did you recompile the `template` test after changing the parameters?

You can recompile by running `./build.sh` in `software/gemmini-rocc-tests`

L-Chocoland, 2022-01-11:
> Did you recompile the `template` test after changing the parameters?
> 
> You can recompile by running `./build.sh` in `software/gemmini-rocc-tests`

yes i recompile it

hngenc, 2022-01-11:
Do you also see this bug when running a _smaller_ config on the FPGA (e.g. 8x8)?

And can you confirm that in your setup, Verilator/VCS works with the 32x32 config?

L-Chocoland, 2022-01-14:
> Do you also see this bug when running a _smaller_ config on the FPGA (e.g. 8x8)?
> 
> And can you confirm that in your setup, Verilator/VCS works with the 32x32 config?
__
I find some error when I build verilator(I confirmed every single code was not changed except the GemminiCustomConfigs.scala ), So I rebuild the chipyard and Gemmini.
the _highPerfInferenceConfig_ in scripts/GemminiCustomConfigs.scala works well on VCU118 and Verilator.
So i think i have met some version problem earlier.
I will test more parameters in future.
this question will close.

closed 2022-01-14 by L-Chocoland

hngenc, 2022-01-15:
Glad to hear! Feel free to re-open the issue if you run into this problem again