ucb-bar/gemmini #184: lean gemmini + low-level ISA problem

thuako, 2022-01-12:
I'm synthesizing and simulating gemmini with firesim. And I get the same error when testing some hardware.

Default gemmini gave no errors, but when gemmini is set to lean gemmini, the simulation never ends when running ResNet-50 with low-level ISA.

I didn't include the assert option in firesim platform config, so I'm not sure exactly why it stopped.

But I got some logs by including firesim print option and perfcount option, and those logs are as follows.
```
# synthesized-prints.out0

CYCLE:   5495742692 Utilization ex q: 16
CYCLE:   5495742692 Utilization st q (incomplete):  0
CYCLE:   5495742692 Utilization ld q (incomplete):  0
CYCLE:   5495742692 Utilization st q:  0
CYCLE:   5495742692 Utilization ld q:  3
CYCLE:   5495742692 Utilization ld q (incomplete):  0
CYCLE:   5495742692 Utilization ex q (incomplete): 15
CYCLE:   5495742692 Utilization st q:  0
CYCLE:   5495742692 Utilization: 19
CYCLE:   5495742692 Utilization ex q (incomplete): 15
CYCLE:   5495742692 Utilization st q (incomplete):  0
CYCLE:   5495742692 Utilization ex q: 16
CYCLE:   5495756586 Utilization ld q:  0
CYCLE:   5495756586 Utilization:  1
CYCLE:   5495756586 Utilization ex q:  1
CYCLE:   5495756586 Utilization st q (incomplete):  0
CYCLE:   5495756586 Utilization ld q (incomplete):  0
CYCLE:   5495756586 Utilization st q:  0
CYCLE:   5495756586 Utilization ld q:  0
CYCLE:   5495756586 Utilization ld q (incomplete):  0
CYCLE:   5495756586 Utilization ex q (incomplete):  0
CYCLE:   5495756586 Utilization st q:  0
CYCLE:   5495756586 Utilization:  1
CYCLE:   5495756586 Utilization ex q (incomplete):  0
CYCLE:   5495756586 Utilization st q (incomplete):  0
CYCLE:   5495756586 Utilization ex q:  1
CYCLE:   5495757586 Utilization ld q:  0
CYCLE:   5495757586 Utilization:  1
```

```
Cycle 5495749999
============================
PerfCounter ex_flush_cycle_FireSim_lazyModule_system_tile_prci_domain_tile_reset_domain_tile_gemmini_ex_controller: 0
PerfCounter ex_mulpre_haz_cycle_FireSim_lazyModule_system_tile_prci_domain_tile_reset_domain_tile_gemmini_ex_controller: 0
PerfCounter ex_preload_haz_cycle_FireSim_lazyModule_system_tile_prci_domain_tile_reset_domain_tile_gemmini_ex_controller: 0
PerfCounter load_dma_wait_cycle_FireSim_lazyModule_system_tile_prci_domain_tile_reset_domain_tile_gemmini_load_controller: 68421478
PerfCounter pooling_cycles_FireSim_lazyModule_system_tile_prci_domain_tile_reset_domain_tile_gemmini_store_controller: 2430209
PerfCounter rdma_active_cycles_FireSim_lazyModule_system_tile_prci_domain_tile_reset_domain_tile_gemmini_spad_reader_core: 82049909
PerfCounter rdma_tl_wait_cycles_FireSim_lazyModule_system_tile_prci_domain_tile_reset_domain_tile_gemmini_spad_reader_core: 63001916
PerfCounter rdma_tlb_wait_cycles_FireSim_lazyModule_system_tile_prci_domain_tile_reset_domain_tile_gemmini_spad_reader_core: 4323286
PerfCounter reservation_station_busy_FireSim_lazyModule_system_tile_prci_domain_tile_reset_domain_tile_gemmini_reservation_station: 191509477
PerfCounter reservation_station_full_FireSim_lazyModule_system_tile_prci_domain_tile_reset_domain_tile_gemmini_reservation_station: 5482429586
PerfCounter st_dma_wait_cycle_FireSim_lazyModule_system_tile_prci_domain_tile_reset_domain_tile_gemmini_store_controller: 43437569
PerfCounter tlb_hits_FireSim_lazyModule_system_tile_prci_domain_tile_reset_domain_tile_gemmini_tlb_tlbs_0: 5127620
PerfCounter tlb_miss_cycles_FireSim_lazyModule_system_tile_prci_domain_tile_reset_domain_tile_gemmini_tlb_tlbs_0_tlb: 5319211186
PerfCounter tlb_reqs_FireSim_lazyModule_system_tile_prci_domain_tile_reset_domain_tile_gemmini_tlb_tlbs_0: 13356929
PerfCounter wdma_active_cycles_FireSim_lazyModule_system_tile_prci_domain_tile_reset_domain_tile_gemmini_spad_writer: 44774870
PerfCounter wdma_tl_wait_cycles_FireSim_lazyModule_system_tile_prci_domain_tile_reset_domain_tile_gemmini_spad_writer: 40635685
PerfCounter wdma_tlb_wait_cycles_FireSim_lazyModule_system_tile_prci_domain_tile_reset_domain_tile_gemmini_spad_writer: 789067

Cycle 5495759999
============================
PerfCounter ex_flush_cycle_FireSim_lazyModule_system_tile_prci_domain_tile_reset_domain_tile_gemmini_ex_controller: 0
PerfCounter ex_mulpre_haz_cycle_FireSim_lazyModule_system_tile_prci_domain_tile_reset_domain_tile_gemmini_ex_controller: 0
PerfCounter ex_preload_haz_cycle_FireSim_lazyModule_system_tile_prci_domain_tile_reset_domain_tile_gemmini_ex_controller: 0
PerfCounter load_dma_wait_cycle_FireSim_lazyModule_system_tile_prci_domain_tile_reset_domain_tile_gemmini_load_controller: 68425041
PerfCounter pooling_cycles_FireSim_lazyModule_system_tile_prci_domain_tile_reset_domain_tile_gemmini_store_controller: 2430209
PerfCounter rdma_active_cycles_FireSim_lazyModule_system_tile_prci_domain_tile_reset_domain_tile_gemmini_spad_reader_core: 82053819
PerfCounter rdma_tl_wait_cycles_FireSim_lazyModule_system_tile_prci_domain_tile_reset_domain_tile_gemmini_spad_reader_core: 63005389
PerfCounter rdma_tlb_wait_cycles_FireSim_lazyModule_system_tile_prci_domain_tile_reset_domain_tile_gemmini_spad_reader_core: 4323313
PerfCounter reservation_station_busy_FireSim_lazyModule_system_tile_prci_domain_tile_reset_domain_tile_gemmini_reservation_station: 191519304
PerfCounter reservation_station_full_FireSim_lazyModule_system_tile_prci_domain_tile_reset_domain_tile_gemmini_reservation_station: 5482439494
PerfCounter st_dma_wait_cycle_FireSim_lazyModule_system_tile_prci_domain_tile_reset_domain_tile_gemmini_store_controller: 43437569
PerfCounter tlb_hits_FireSim_lazyModule_system_tile_prci_domain_tile_reset_domain_tile_gemmini_tlb_tlbs_0: 5127784
PerfCounter tlb_miss_cycles_FireSim_lazyModule_system_tile_prci_domain_tile_reset_domain_tile_gemmini_tlb_tlbs_0_tlb: 5319211472
PerfCounter tlb_reqs_FireSim_lazyModule_system_tile_prci_domain_tile_reset_domain_tile_gemmini_tlb_tlbs_0: 13357286
PerfCounter wdma_active_cycles_FireSim_lazyModule_system_tile_prci_domain_tile_reset_domain_tile_gemmini_spad_writer: 44774870
PerfCounter wdma_tl_wait_cycles_FireSim_lazyModule_system_tile_prci_domain_tile_reset_domain_tile_gemmini_spad_writer: 40635685
PerfCounter wdma_tlb_wait_cycles_FireSim_lazyModule_system_tile_prci_domain_tile_reset_domain_tile_gemmini_spad_writer: 789067

Cycle 5495769999
============================
PerfCounter ex_flush_cycle_FireSim_lazyModule_system_tile_prci_domain_tile_reset_domain_tile_gemmini_ex_controller: 0
PerfCounter ex_mulpre_haz_cycle_FireSim_lazyModule_system_tile_prci_domain_tile_reset_domain_tile_gemmini_ex_controller: 0
PerfCounter ex_preload_haz_cycle_FireSim_lazyModule_system_tile_prci_domain_tile_reset_domain_tile_gemmini_ex_controller: 0
PerfCounter load_dma_wait_cycle_FireSim_lazyModule_system_tile_prci_domain_tile_reset_domain_tile_gemmini_load_controller: 68425041
PerfCounter pooling_cycles_FireSim_lazyModule_system_tile_prci_domain_tile_reset_domain_tile_gemmini_store_controller: 2430209
PerfCounter rdma_active_cycles_FireSim_lazyModule_system_tile_prci_domain_tile_reset_domain_tile_gemmini_spad_reader_core: 82053819
PerfCounter rdma_tl_wait_cycles_FireSim_lazyModule_system_tile_prci_domain_tile_reset_domain_tile_gemmini_spad_reader_core: 63005389
PerfCounter rdma_tlb_wait_cycles_FireSim_lazyModule_system_tile_prci_domain_tile_reset_domain_tile_gemmini_spad_reader_core: 4323313
PerfCounter reservation_station_busy_FireSim_lazyModule_system_tile_prci_domain_tile_reset_domain_tile_gemmini_reservation_station: 191529304
```

Note : I changed ReservationStation.scala as follows.
```
  val cntr = Counter(2000000)
  when (cntr.inc() || ((cycles_since_issue % 1000.U) === 0.U && cycles_since_issue > 0.U)  ) {
    printf(p"Utilization: $utilization\n")
    printf(p"Utilization ld q (incomplete): $utilization_ld_q_unissued\n")
    printf(p"Utilization st q (incomplete): $utilization_st_q_unissued\n")
    printf(p"Utilization ex q (incomplete): $utilization_ex_q_unissued\n")
    printf(p"Utilization ld q: $utilization_ld_q\n")
    printf(p"Utilization st q: $utilization_st_q\n")
    printf(p"Utilization ex q: $utilization_ex_q\n")

    if (use_firesim_simulation_counters) {
      printf(SynthesizePrintf("Utilization: %d\n", utilization))
      printf(SynthesizePrintf("Utilization ld q (incomplete): %d\n", utilization_ld_q_unissued))
      printf(SynthesizePrintf("Utilization st q (incomplete): %d\n", utilization_st_q_unissued))
      printf(SynthesizePrintf("Utilization ex q (incomplete): %d\n", utilization_ex_q_unissued))
      printf(SynthesizePrintf("Utilization ld q: %d\n", utilization_ld_q))
      printf(SynthesizePrintf("Utilization st q: %d\n", utilization_st_q))
      printf(SynthesizePrintf("Utilization ex q: %d\n", utilization_ex_q))
    }
```

I am wondering if you have also encountered this problem, what is the cause and how to fix it.

thuako, 2022-01-12:
Additionally, if the same hardware and the same binary(low-level ISA) are simulated on the midas verilator platform, it seems to work without any problem.

hngenc, 2022-01-14:
I'm building the `leanConfig` in FireSim now to try to reproduce this.

Just so I'm clear, are these the steps you followed?
* Build the `leanConfig` in Firesim
* Run `resnet50-linux` on the `leanConfig` in FireSim
* Observe that `resnet50-linux` never finishes, and never prints anything

thuako, 2022-01-15:
> Build the leanConfig in Firesim
> Run resnet50-linux on the leanConfig in FireSim
> Observe that resnet50-linux never finishes, and never prints anything

Yes, I followed these the steps.

Additionally, I builded and tested various `leanConfig` Gemmini with different CPU and L2 cache.

```
class LeanGemminiRocketConfig extends Config(
  new gemmini.DefaultGemminiConfig(gemmini.GemminiConfigs.leanConfig) ++
  new freechips.rocketchip.subsystem.WithNBigCores(1) ++
  new chipyard.config.AbstractConfig)

class LeanGemmini2MBRocketConfig extends Config(
  new gemmini.DefaultGemminiConfig(gemmini.GemminiConfigs.leanConfig) ++
  new freechips.rocketchip.subsystem.WithNBigCores(1) ++
  new freechips.rocketchip.subsystem.WithInclusiveCache(capacityKB=2048) ++
  new chipyard.config.AbstractConfig)

class LeanGemminiLargeBoomConfig extends Config(
  new gemmini.DefaultGemminiConfig(gemmini.GemminiConfigs.leanConfig) ++
  new boom.common.WithNLargeBooms(1) ++
  new chipyard.config.AbstractConfig)
```

I also tested this hardware with various `resnet50-linux` configured as low level ISA but using different double buffer options and printing the current I, J, K tile numbers.

and the results are below.

| hardware                   | loop ISA | low-level ISA | low-level ISA w/ print IJK | low-level ISA with double buffer |
|----------------------------|----------|---------------|----------------------------|----------------------------------|
| leanConf + Rocket          |     o    |       x       |              o             |                 x                |
| leanConf + Boom            |     o    |       x       |              x             |                 x                |
| leanConf + Rocket + 2MB L2 |     o    |       x       |              x             |                 x                |

hngenc, 2022-01-15:
Sorry, can you explain what the difference is between your `loop ISA` tests and `low-level ISA` tests?

Did you change `gemmini.h` to run the `low-level ISA` tests?

thuako, 2022-01-15:
Oh, sorry. I just named the binaries which compiled with default `gemmini.h` to `loop ISA` due to default gemmini.h using `gemmini_loop_ws` and `gemmini_loop_conv_ws` ISA.

`low-level ISA` is just uncomment code in the `sp_tiled_matmul_ws` and `sp_tiled_conv` for using low-level ISA, and I don't any change in gemmini.h .

`low-level ISA w/ print` is I insert the printf into sp_tiled_matmul_ws as below.

```
  // Move-in D
  if (D != NULL && !no_bias) {
    for (size_t i = 0; i < I; i++) {
      const size_t rows = DIM - (i == I-1 ? pad_I : 0);
      for (size_t j = 0; j < J; j += D_blocks) {
        const size_t bias_row = repeating_bias ? 0 : i;
        const void * const D_dram_addr = (int8_t *)D + (bias_row * D_row_stride + j)*DIM*sizeof_D;
        const uint32_t D_sp_addr_acc = D_sp_addr_start + (i*J + j)*DIM;
        size_t blocks = j + D_blocks <= J ? D_blocks : J-j;
        const size_t cols = blocks * DIM - (j + blocks >= J ? pad_J : 0);
        gemmini_extended_mvin3(D_dram_addr, D_sp_addr_acc, cols, rows);
      }
    }
  }

  for (size_t j = 0; j < J; j++) {
    for (size_t k = 0; k < K; k++) {
      for (size_t i = 0; i < I; i++) {
        //printf("%d,%d,%d\n",i, k, j);
        const uint32_t A_sp_addr = a_transpose ? (A_sp_addr_start + (k*I + i)*DIM) :
          (A_sp_addr_start + (i*K + k)*DIM);
        const uint32_t B_sp_addr = b_transpose ? (B_sp_addr_start + (j*K + k)*DIM) :
          (B_sp_addr_start + (k*J + j)*DIM);
        const uint32_t C_sp_addr = C_sp_addr_start + (i*J + j)*DIM;

        // Mvin A
        if (a_transpose) {
          if (j == 0 && i % A_blocks == 0) {
            const elem_t * const A_dram_addr = A + (k*A_row_stride + i)*DIM;
            const size_t blocks = i + A_blocks <= I ? A_blocks : I-i;
```

hngenc, 2022-01-15:
Could you share a copy of your "low-level ISA `gemmini.h`". You could save it in a [Gist](https://gist.github.com/).

I think I know what changes you made, but I just want to be sure I'm running the same thing

thuako, 2022-01-15:
This is my gemmini.h [https://gist.github.com/thuako/ff419c06ce3eaa5e8292532d307cdca4](url)

Thank you for your interest in this issue.

hngenc, 2022-01-15:
I'm able to replicate this issue. I'll try to figure out the cause of this bug, though this won't be very high-priority, since the loop-ISA seems to be working fine.

thuako, 2022-01-16:
I started developing on top of gemmini to create NPUs for specific workloads because gemmini is the most versatile and compatible when comparing other NPU platforms.

But the NPU I made seems to have the same bugs as leanConfig , even though I haven't changed max flight requests. (There is no bug on midas platform, but on firesim it is stopped by gemmini_timeout assert. And if I put printf between the loops, it works fine.)

In the end, I'm making LoopMatmul and LoopConv dedicated to specific workloads to solve the CPU bottleneck and the bug above.

However, as #161 , loop-ISA has some drawbacks in terms of versatility.

So, even if this issue isn't a high priority right now, I think it should be resolved for many people to use Gemmini.

Thank you for your response!

hngenc, 2022-01-19:
Quick update:

I'm still debugging this. The error seems to be caused by a `preload` instruction which is issued to from the `ReservationStation` to the `ExecuteController`, but is never marked as "complete" by the `ExecuteController`.

This seems to be very timing-dependent, which means I haven't been able to replicate this on VCS or MIDAS. Debugging on FireSim is a lot slower than debugging on VCS/MIDAS, but I'm still making progress.

thuako, 2022-01-20:
Thank you for giving me updates.

And I have a one question. Is it common for bugs to occur only in firesim?

hngenc, 2022-01-20:
Not very common.

In this case, I believe the bug is occurring because OS interrupts are exposing a bug in the ReservationStation.

With Verilator/MIDAS, we're not running Linux, so these interrupts don't happen.

hngenc, 2022-01-21:
This bug should have been fixed by https://github.com/ucb-bar/gemmini/pull/189, which has been merged into the `dev` branch.

I'm closing this issue for now, but if you find that it still persists even on the `dev` branch, then feel free to re-open it.

commit 5d85c726 (2022-01-21): Fix Unpredictable Infinite Stall Caused By OS Interrupts (#189)

Fixes #184
closed 2022-01-21 by hngenc