ucb-bar/gemmini #188: [Question] Can gemmini work with (meshrow, meshcol, tilerow, tilecol)=(4, 4, 4, 4)?

rgb000000, 2022-01-20:
Hello,
I find that there are two extreme cases in [the gemmini paper](https://people.eecs.berkeley.edu/~ysshao/assets/papers/genc2021-dac.pdf), TPU-like and NVDLA-like.
In gemmini project, default config is `(meshrow, meshcol, tilerow, tilecol)=(16, 16, 1, 1)`, I think it is in TPU-like mode.
I try to use `(meshrow, meshcol, tilerow, tilecol)=(4, 4, 4, 4)` to find out whether gemmini can work in a intermediate state. 
But it can't work because [this assert](https://github.com/ucb-bar/gemmini/blob/master/src/main/scala/gemmini/Util.scala#L12).

So, can't gemmini work under this case?

hngenc, 2022-01-21:
Thanks for identifying this bug.

This is fixed by #190, which has now been merged into the `dev` branch. The "intermediate" designs (like your 4x4x4x4 design) should work fine now.

Feel free to re-open the issue if you still run into this error in the `dev` branch.

closed 2022-01-21 by hngenc

8u9mak3r, 2024-03-16:
> Thanks for identifying this bug.
> 
> This is fixed by #190, which has now been merged into the `dev` branch. The "intermediate" designs (like your 4x4x4x4 design) should work fine now.
> 
> Feel free to re-open the issue if you still run into this error in the `dev` branch.

@hngenc But I'm now facing another problem. if you once changed the configs to (meshrow, meshcol, tilerow, tilecol)=(16, 16, 1, 1) or (4,4,4,4) and now change it to (1,1,16,16) and build the verilator, the simulator built will apply the previous configs you set however.Is this an internal bug or Gemmini simply cannot work under this case?

commit 96011d17 (2022-01-21): Fix incorrect assertions from firing when building non-systolic configs (#190)

Fixes #188