ucb-bar/gemmini #193: Testing and Performance Break-down for Gemmini

ChaoGaoUCR, 2022-02-01:
Hi, guys,

Sorry to bother you here,
I am trying to implement **one iteration of Feed Forward and loss function backward** to do a performance breakdown experiments to see the latency and memory access performance, I have a few questions about it:
0. Is there any specific tutorial about integrated Gemmini-Rocketchip system to FireSim (I didn't find that in the IISWC tutorial)
1. If I use Spike simulation, can I get the DRAM performance? If so, when I try to change the size of Systolic Array, how should I reset the parameter in Spike simulator, if not, does that mean I have to use FireSim?
2. I saw some examples in docker provided in IISWC like alexnet.c/resnet-50.c which used middle-level function, if I want to do the backward test, does that mean I have to write the c file like that and change the reading manner of weight tensor? Also, in that case, how should I off-load the pooling layer backward function to CPU and how do I keep the pooling layer index with ROCC system? (I found no middle-level function support for training in gemmini.h file)

Thanks for your patience
Chao

hngenc, 2022-02-01:
>  Is there any specific tutorial about integrated Gemmini-Rocketchip system to FireSim

We don't have any tutorial specifically related to simulating Gemmini on FireSim. But the [FireSim docs](https://docs.fires.im/en/latest/) are pretty up-to-date, and they should have enough to get you started.

My recommendation would be to go through the *Getting Started* section of the FireSim docs. Afterwards, you can try following the steps below to run Gemmini workloads in particular:

1. Edit `gemmini/software/overlay/root/run-tests.sh`. You can decide exactly which Gemmini binaries you want to run in that script. At the end of the script, just remember to run `poweroff -f` to shut down the simulation.

2. Afterwards, you need to build the “Firemarshal workload”. This is the step that actually creates a filesystem with all your binaries that will be copied over to the simulated Gemmini accelerator. To complete this step, run the following commands:
```
cd ~/firesim/sw/firesim-software

./marshal -v --workdir /home/centos/firesim/target-design/chipyard/generators/gemmini/software build gemmini-tests.json &&
  ./marshal -v --workdir /home/centos/firesim/target-design/chipyard/generators/gemmini/software install gemmini-tests.json
```

3. Afterwards, you need to build the FPGA image of the Gemmini accelerator. Go to `~/firesim/deploy/config_build_recipes.ini`, and search for this line:

https://github.com/firesim/firesim/blob/master/deploy/sample-backup-configs/sample_config_build_recipes.ini#L55

I would recommend changing the `F110MHz` to `F30MHz`. That will soon be the default anyways

4. Afterwards, open up `~/firesim/deploy/config_build.ini`, and add `firesim-rocket-singlecore-gemmini-no-nic-l2-llc4mb-ddr3` underneath the `[builds]` section. I would also recommend deleting any other builds in that section. Finally, I would recommend changing the `buildinstancemarket=ondemand` line to `buildinstancemarket=spot` to save money.

5. Next, go to `~/firesim` and run `firesim buildafi`. *This step may take many hours to complete.*

6. After the build completes, the output will give you some lines of text that you need to add to `deploy/config_hwdb.ini`. The text will look something like this:

```
[firesim-rocket-singlecore-gemmini-no-nic-l2-llc4mb-ddr3]
agfi=agfi-009b545d2276a3681
deploytripletoverride=None
customruntimeconfig=None
```

Delete the old `[firesim-rocket-singlecore-gemmini-no-nic-l2-llc4mb-ddr3]` entry in `deploy/config_hwdb.ini`, and replace it with these new lines.

7. Next, update `deply/config_runtime.ini` by setting the `defaulthwconfig` to `firesim-rocket-singlecore-gemmini-no-nic-l2-llc4mb-ddr3`. Also, set the `workloadname` to `gemmini-tests.json`.

8. Finally, run these these steps to spin up an F1 instance, run your test, and shut down the F1 instance:

```
# firesim launchrunfarm &&
    firesim infrasetup &&
    firesim runworkload # &&
    # firesim terminaterunfarm --forceterminate
```

> If I use Spike simulation, can I get the DRAM performance?

Unfortunately not. Spike doesn't accurately model DRAM performance.

> I saw some examples in docker provided in IISWC like alexnet.c/resnet-50.c which used middle-level function, if I want to do the backward test, does that mean I have to write the c file like that and change the reading manner of weight tensor?

If you want to run training workloads on Gemmini, I would recommend using [ONNX Runtime](https://github.com/ucb-bar/onnxruntime-riscv/), which should be able to figure out the right data formats and functions to run for you.

If you want to implement training in the C file directly, then you'll need to use special arguments in `tiled_conv`:

https://github.com/ucb-bar/gemmini-rocc-tests/blob/dev/include/gemmini.h#L2155-L2157

These arguments can be used to implement transpose convolutions for backward pass of convolutions.

However, Gemmini doesn't currently have a good solution for the backward pass on batch norms, or for the backward pass of max-pooling. If you're running training, then max-pooling needs to happen on the CPU, so that you know the indices of the max-pooled elements (as you yourself pointed out).

Hope that helps!

ChaoGaoUCR, 2022-02-01:
Thank you so much, Hasan!

Thank you very much for your specific detailed tutorial, Hasan!
I really appreciate your support and hope I can give you some good feedback later!

Cheers!
Chao

hngenc, 2022-03-07:
Closing this issue for now, but feel free to open it again if you run into any issues

closed 2022-03-07 by hngenc

aladerran, 2023-11-12:
@hngenc Hi! A quick question. I am also trying to use firesim to run onnx model. My question is after I make changes to GemminiCustomConfig.scala, would the new bitstream (firesim-rocket-singlecore-gemmini-no-nic-l2-llc4mb-ddr3 in this case) I get from ```firesim buildbitstream``` be updated with the new config?

I want to run onnx model with "trainingConfig" or even some custom configs, I wonder if it will work?

Thanks in advance!!!

hngenc, 2023-11-30:
@aladerran You'll want to check in your `config_build_recipes` file to make sure that `firesim-rocket-singlecore-gemmini-no-nic-l2-llc4mb-ddr3` corresponds to `CustomGemminiSoCConfig`. By default; I don't believe that it does (which is probably an oversight that we should fix); instead, it corresponds to `GemminiRocketConfig`. If you update your `config_build_recipes` so that it instead corresponds to `CustomGemminiSoCConfig`, then I believe it should work (but let me know if it doesn't).

aladerran, 2023-12-03:
@hngenc Thanks so much for you response. I will give it a try and update any useful outcome here :D

aladerran, 2024-02-03:
@hngenc Hi Hasan,

I changed the ```config_build_recipes``` and make it correspond to  ```CustomGemminiSoCConfig```, and I successfully built a desired bitstream. However, when I try to enlarge the capacity of Spad and Acc, or when increasing the number of CPU to quad-core (works fine w/ dual-core), I met ```Resource (over-)Utilization``` problem, which I cannot find related issues on firesim or chipyard websites. Do you have any suggestions? 

```
firesim_custom_gemmini_soc_no_nic_l2_llc4mb_ddr3:
    PLATFORM: f1
    TARGET_PROJECT: firesim
    DESIGN: FireSim
    TARGET_CONFIG: DDR3FRFCFSLLC4MB_WithDefaultFireSimBridges_WithFireSimTestChipConfigTweaks_chipyard.CustomGemminiSoCConfig
    PLATFORM_CONFIG: WithAutoILA_BaseF1Config
    deploy_quintuplet: null
    platform_config_args:
        fpga_frequency: 30 # AJG: conservative for now, later sweep for higher frequency
        build_strategy: TIMING
    post_build_hook: null
    metasim_customruntimeconfig: null
    bit_builder_recipe: bit-builder-recipes/f1.yaml
```

```
class CustomQuadBoomGemminiSoCConfig extends Config(
  new gemmini.GemminiCustomConfig ++

  // Set your custom L2 configs
  new chipyard.config.WithL2TLBs(512) ++

  new freechips.rocketchip.subsystem.WithInclusiveCache(
    nWays = 8,
    capacityKB = 512,
    outerLatencyCycles = 40,
    subBankingFactor = 4
  ) ++

  // Set the number of CPUs you want to create
  new chipyard.CustomGemmminiCPUConfigs.CustomCPU(4) ++

  new chipyard.config.WithSystemBusWidth(GemminiCustomConfigs.customConfig.dma_buswidth) ++
  new chipyard.config.AbstractConfig
)
```

Error log: 
[2024-02-03--02-29-25-buildbitstream-XGMR35ECBNDGHYS3.log](https://github.com/ucb-bar/gemmini/files/14150487/2024-02-03--02-29-25-buildbitstream-XGMR35ECBNDGHYS3.log)

This is kinda urgent since I am trying to do some DSE on Gemmini. Looking forward to your reply!

hngenc, 2024-02-03:
> I met Resource (over-)Utilization problem, which I cannot find related issues on firesim or chipyard websites

If you're struggling to fit the hardware on the FPGA, then you might want to look into using Gemmini's [`dummyConfig`](https://github.com/ucb-bar/gemmini/blob/dev/src/main/scala/gemmini/Configs.scala#L167). The Dummy Config replaces all the arithmetic operations (and many of the SRAMs) with no-ops, which greatly reduces its area consumption.

Functionally, the Dummy Config won't return correct results for matrix multiplications, but it will achieve the exact same cycle-accurate performance as Gemmini's default config. So if you only want to evaluate Gemmini's performance, then it might be sufficient for you.

aladerran, 2024-02-03:
@hngenc Thanks very much for your response. I will try it :)