ucb-bar/gemmini #196: A issue in the script

BrianQian1999, 2022-02-15:
I was trying to generate the waveform: 
```
./script/run-verilator.sh --debug template 
```

There would be an error: 
```
./scripts/run-verilator.sh: line 91: ./simulator-chipyard-CustomGemminiSoCConfig-debug: No such file or directory
```

The problem seems to lie in the script: 
```
if [ $debug -eq 1 ]; then
    DEBUG="-debug -v ${ROOT}${WAVEFORM}"
else
    DEBUG=""
fi

cd ../../sims/verilator/
./simulator-chipyard-CustomGemminiSoCConfig${DEBUG} $PK ${full_binary_path}
```

BrianQian1999, 2022-02-15:
Sorry, I didn't successfully build verilator with debug.

closed 2022-02-15 by BrianQian1999

hngenc, 2022-02-15:
Which commit are you on?

We fixed an issue with the debug scripts on dev recently. (But we haven't merged it to master yet).

You don't need to rebuild everything with the dev branch, but could you try replacing your `build-verilator.sh` script with the one on Gemmini's `dev` branch, and rebuilding the Verilator simulator? That might fix your problem.

BrianQian1999, 2022-02-15:
In dev branch, the $build-verilator.sh$ is trying to build: 
```
cd ../../sims/verilator/
make ${debug} CONFIG=CustomGemminiSoCConfig
```
However, $run-verilator.sh$，is trying to call:
```
cd ../../sims/verilator/
./simulator-chipyard-CustomGemminiSoCConfig${DEBUG} $PK ${full_binary_path}
```
while $ ${DEBUG} $ is 
```
if [ $debug -eq 1 ]; then
    DEBUG="-debug -v ${ROOT}${WAVEFORM}"
else
    DEBUG=""
fi
```

which will make us call $simulator-chipyard-CustomGemminiSoCConfig-debug$, and it doesn't exist.

BrianQian1999, 2022-02-15:
In ``dev`` branch, the ``build-verilator.sh`` would build:
```
cd ../../sims/verilator/
make ${debug} CONFIG=CustomGemminiSoCConfig
```
However, ``run-verilator.sh`` would run: 
```
cd ../../sims/verilator/
./simulator-chipyard-CustomGemminiSoCConfig${DEBUG} $PK ${full_binary_path} 
```

i.e., it would call ``simulator-chipyard-CustomGemminiSoCConfig${DEBUG}`` while 
```
DEBUG="-debug -v ${ROOT}${WAVEFORM}"
```
which means it would try to call ``simulator-chipyard-CustomGemminiSoCConfig-debug``, and it does not exist.

hngenc, 2022-02-15:
Yes, but `make ${debug} CONFIG=CustomGemminiSoCConfig` _should_ build `simulator-chipyard-CustomGemminiSoCConfig-debug`. At least on the dev branch. And if it doesn't, that's a bug that we need to fix.

hngenc, 2022-02-16:
No problem, thanks for confirming!

kahretsinanil, 2022-05-10:
> Yes, it generates the right simulator with `-debug` after rebuilding. Thank u for answering~

hi,
how did you solve this problem? where did you add the ```-debug``` ?