ucb-bar/gemmini #197: Gemmini hardware

BrianQian1999, 2022-02-16:
Hi, 

I'm trying to synthesize the generated SoC in verilog RTL
`` chipyard.TestHarness.CustomGemminiSoCConfig.top.v``

Is ``Chiptop`` the top module of the whole SoC? And does it include the DRAM model?

hngenc, 2022-02-16:
Yes, ChipTop represents the whole chip. I don't believe it includes the DRAM model.

You might also want to look at Chipyard's VLSI flow docs:

https://chipyard.readthedocs.io/en/latest/VLSI/index.html

BrianQian1999, 2022-02-20:
I tried to utilize Hammer but found that it's too hard to build chipyard on the EDA server provided by my school (the OS and software kits are too old). Thus I moved to my own flow, but encountered another problem. 

I'm trying to figure out which verilog/systemverilog file does verilator use when running simulation, but cannot find the testbench under ``generated-src/chipyard.TestHarness.CustomGemminiSoCConfig`` directory. 

According to the generated VCD file, there should be a module named ``TOP`` corresponds to the testbench, while ``chiptop`` was the DUT.

BrianQian1999, 2022-02-24:
Hi, 

I was trying to simulate some DNN workload in Verilator and combine the generated VCD file to my own VLSI flow.   
I ran mobilenet, which generated ~800GB VCD file. This is too large for my EDA tool. 
Thus I'd like to deploy some smaller model, but got confused how do you write or generate the parameters of NN in C, 
e.g., `software/gemmini-rocc-tests/imagenet/mobilenet_params.h` 
Currently I wish to generate `alexnet_params.h`

hngenc, 2022-02-24:
Running MobileNet or ResNet50 in Verilator/VCS is not recommended, because the simulations take too long to complete.

We instead recommend running them on FireSim, which uses FPGAs to simulate the hardware.

As for the params file, we used a Python script to convert the weights in PyTorch models into C header files. But those scripts were really messy, and were never open-sourced.

closed 2022-03-02 by BrianQian1999