ucb-bar/gemmini #199: Spike FAIL while Verilator PASS

YingWu321, 2022-02-22:
Hi,

I changed some parts of the gemmini project and used the provided scripts to build. 

Because Spike runs much faster, I run all test cases using it. But there are 4 files not passing including padded-baremetal, aligned-barematal, mvin_mvout-baremetal and mvin_mvout_block_stride-baremetal.  All of them failed due to inequality between input and output. The following picture shows the result of aligned-baremetal(Spike).
[image]

Then I tried running these files with Verilator and they all passed.

I'm bothering which one is correct.

hngenc, 2022-02-22:
Which parts did you change?

Looking at your output, it seems that Spike is using signed 8-bit integers, but Verilator is using *unsigned* 8-bit integers. Did you run `scripts/build-spike.sh` after making your changes?

hngenc, 2022-02-23:
Oh, I see.

Unfortunately, changes to the RTL aren't reflected in Spike. Spike is written in C++, and it doesn't know what your Chisel or Verilog changes are.

Verilator/VCS should work with your latest changes though.

closed 2022-03-03 by hngenc