ucb-bar/gemmini #200: Customizing Hardware

uditagarwal97, 2022-02-23:
Hi,

I've made some custom functional changes in the PE. However, even after rebuilding Gemmini's hardware, the existing test cases are not reflecting the changes I made. 
Here's what I did:
1. Made changes in PE.scala file
2. Rebuild Verilator and Spike via _scripts/build-verilator.sh_ and _scripts/build-spike.sh_ scripts.
3. Rebuild gemmini-rocc-tests
4. Run test cases via scripts/run-spike.sh script (e.g. ./scripts/run-spike.sh conv)

The test cases are not reflecting the changes I made. For example, if in the PE.scala file, I set the output of every PE equal to zero, then I expect the output of the conv or matmul operator to 0 always, irrespective of the inputs. However, even after making this change, the test cases (conv and imagenet) are still passing. I'm just wondering If I'm missing any step while rebuilding the Gemmini hardware?

I'm using the latest version of Gemmini (master branch).

hngenc, 2022-02-23:
Changes to the RTL will not be reflected in Spike simulations, because Spike is only a functional, instruction-level simulator written in C++.

But if you change PE.scala, you should be able to see the difference in Verilator or VCS simulations.

uditagarwal97, 2022-02-23:
Thanks, that worked out!

closed 2022-02-23 by uditagarwal97

hngenc, 2022-02-23:
Great, thanks for confirming.

Also, if you do want to update Spike, then you can modify the source code of the `riscv-isa-sim` program. It should be somewhere in Chipyard.