ucb-bar/gemmini #201: make error when run build-vcs.sh

BrianQian1999, 2022-03-02:
I ran `./scripts/build-vcs.sh --debug` and got:
```
make[1]: Entering directory '/home/brianqian/chipyard/sims/vcs/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.debug'
g++ -w  -pipe -fPIC -O3 -I/home/brianqian/synopsys/vcs/O-2018.09-SP2/include    -c /home/brianqian/chipyard/sims/vcs/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/mm.cc
g++ -w  -pipe -fPIC -O3 -I/home/brianqian/synopsys/vcs/O-2018.09-SP2/include    -c /home/brianqian/chipyard/sims/vcs/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/mm_dramsim2.cc
In file included from /home/brianqian/chipyard/sims/vcs/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/mm_dramsim2.cc:3:
/home/brianqian/chipyard/sims/vcs/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/mm_dramsim2.h:7:10: fatal error: DRAMSim.h: No such file or directory
    7 | #include <DRAMSim.h>
      |          ^~~~~~~~~~~
compilation terminated.
make[2]: *** [filelist:11: mm_dramsim2.o] Error 1
make[1]: *** [Makefile:104: product_clean_order] Error 2
make[1]: Leaving directory '/home/brianqian/chipyard/sims/vcs/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.debug'

```
Which maybe was because of linkage when compiling.

hngenc, 2022-03-03:
1. Which Gemmini commit are you on?
2. Can you confirm that all the commands [over here](https://github.com/ucb-bar/gemmini#installing-chipyard-and-spike) ran correctly?
3. Can you confirm that you ran `source env.sh` before running the `make` command?

BrianQian1999, 2022-03-04:
1. I was on `master` commit, and also tried the updated scripts on `dev` commit. 
2. Yes. 
3. Yes.

closed 2022-03-04 by BrianQian1999

hngenc, 2022-03-05:
@BrianQian1999 Can you share how you resolved this issue, just in case someone in the future also runs into it?

BrianQian1999, 2022-03-05:
As `DRAMsim.h` should be included, 
```
g++ -w  -pipe -fPIC -O3 -I/home/brianqian/synopsys/vcs/O-2018.09-SP2/include    -c /home/brianqian/chipyard/sims/vcs/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/mm.cc
```
I just simply copy DRAMsim.h to `/home/brianqian/synopsys/vcs/O-2018.09-SP2/include`, along with the other needed header file.

hngenc, 2022-03-05:
I see, thanks for debugging that!