ucb-bar/gemmini #202: Where is the Verilog adapter?

MrJbriggs, 2022-03-02:
It supposedly is the bridge between my Gemmini verilog mesh and the Chisel interface.  Where to find it?

hngenc, 2022-03-03:
I think you're referring to the Gemmini Verilog mesh we created for our Hardware-For-Machine-Learning class.

You can find the Verilog mesh [over here](https://github.com/ucb-bar/gemmini/blob/ee290-sp21-lab2/src/main/resources/vsrc/MeshBlackBox.v).

You may also want to use the `ee290-sp21-lab2` branch if you're trying to follow along with the class.

closed 2022-03-03 by hngenc

MrJbriggs, 2022-03-03:
Thank you, @hngenc !  This is a big help

I had made up my own mesh Bbox verilog, since one did not get populated. Had no idea the Adapter was also a verilog.

So just clone that branch and I should be golden as far as getting Lab 2 (and Lab 3) done?

Thank you

hngenc, 2022-03-03:
> So just clone that branch and I should be golden as far as getting Lab 2 (and Lab 3) done?

I hope so, but the branch is a bit old, so I'm not 100% sure if everything still works tbh

MrJbriggs, 2022-03-03:
Haha, ok. 

I'll first try plugging the needed files from the course branch into my existing Chipyard and see how far I get with it. Thanks again

hngenc, 2022-03-03:
No problem :)

MrJbriggs, 2022-03-04:
@hngenc

UPDATE:
Not surprisingly, doing this build did not create the bare-metal tests needed for Lab 2 since those are not part of the official release.

cd chipyard/generators/gemmini/software/gemmini-rocc-tests
./build.sh

Some follow-up questions below:

**Q1**. Is what I have below the correct sequence of steps to use ee290-sp21-lab2 branch in my existing chipyard? It is based on what you have under "Installing Chipyard and Spike". That line in bold is not part of the instructions on the Chipyard github, what is it for? I also noticed there is another branch called ee290 in the dropdown menu, should I care?  

git clone https://github.com/ucb-bar/chipyard.git
cd chipyard
**git checkout 117624d8eea27bafd613eec09e9b9b3e31239e08**
./scripts/init-submodules-no-riscv-tools.sh
./scripts/build-toolchains.sh esp-tools

source env.sh

cd generators/gemmini
git fetch && git checkout -b ee290-sp21-lab2 https://github.com/ucb-bar/chipyard.git
git submodule update

**Q2.** Why is this not part of the setup instructions in ee290?

cd chipyard/generators/gemmini
./scripts/setup-paths.sh

**Q3**
Abe Gonzalez mentioned that there is a HW tiler that would need to be disabled for this Lab. I assume that I'll get that modification in this branch: ee290-sp21-lab2 as well?

Thank you Hasan

hngenc, 2022-03-04:
You need to run `git submodule update` to get the right baremetal tests.

The newer build scripts (which includes `setup-paths.sh`) was added after we created the lab.

MrJbriggs, 2022-03-04:
Thanks @hngenc 

Since I already have Chipyard cloned, so just do this then (I am a Github newbie)?:

`cd generators/gemmini
git fetch && git checkout -b ee290-sp21-lab2 https://github.com/ucb-bar/gemmini.git
git submodule update`

MrJbriggs, 2022-03-04:
@hngenc 

I tried to populate this branch: https://github.com/ucb-bar/gemmini/tree/ee290-sp21-lab2 to see if it will work with the existing Chipyard (Master):

I did:
mv gemmini gemmini.old
mkdir gemmini
cd gemmini
git fetch
git checkout https://github.com/ucb-bar/gemmini/tree/ee290-sp21-lab2

I got this message:
error: pathspec 'https://github.com/ucb-bar/gemmini/tree/ee290-sp21-lab2' did not match any file(s) known to git

Any clue?  Thank you

MrJbriggs, 2022-03-04:
@hngenc 

I tried this:

mv gemmini gemmini.old
mkdir gemmini
cd generators/gemmini
git fetch && git checkout -b ee290-sp21-lab2
git submodule update

It asked for a username and password. It didn't like mine :) Seems like it is trying to access Enterprise?

hngenc, 2022-03-05:
Hmm, it's probably trying to access a sub-repo that was only shared with the students that semester.

Maybe your best bet would be to look at the branch to see how to create your own Verilog mesh, but do your actual work on the `master` or `dev` branch of Gemmini?