ucb-bar/gemmini #203: How to configure chipyard/Gemmini to see a and use my verilog mesh?

MrJbriggs, 2022-03-02:
I have the verilog located in vsrc like this:

generators/yourproject/
    build.sbt
    src/main/
        scala/
        resources/
            vsrc/
                YourFile.v

But what else is needed? What does this instruction in section 6.10.1 of Chipyard doc mean?:

```
lazy val tapeout = conditionalDependsOn(project in file("./tools/barstools/tapeout/"))
  .dependsOn(chisel_testers, example, yourproject)
  .settings(commonSettings)
```

MrJbriggs, 2022-03-03:
Getting some traction with this:

[https://www.chisel-lang.org/chisel3/docs/explanations/blackboxes.html#:~:text=Chisel%20BlackBoxes%20are%20used%20to,IP%20not%20defined%20in%20Chisel.](url)

And this

[https://inst.eecs.berkeley.edu/~cs250/sp16/disc/Blackboxing.pdf](url)

hngenc, 2022-03-03:
You can use the `ee290-sp21-lab2` branch for an example of how to incorporate Verilog into your design

closed 2022-03-03 by hngenc

MrJbriggs, 2022-03-03:
Got it. Thank you @hngenc

I really appreciate your help