ucb-bar/gemmini #204: Installing Chipyard and Spike

minmin20028, 2022-03-04:
Hi. I'm new to chipyard and gemmini systems.
Currently, I'm following the Installing Chipyard and Spike section of the Quick start.

Among them, **./scripts/build-toolchains.sh esp-tools**
I doubt that this part was executed properly.

This is because env.sh file was created, but env-esp-tools.sh was not created and not properly linked.

Also, if follow the process further,

**cd -
cd toolchains/esp-tools/riscv-isa-sim/build**

In this part, the directory called build does not exist at all.

It's probably an out of sync problem with the chipyard repository, and I'm just not sure how to fix it.

Any help would be greatly appreciated.

hngenc, 2022-03-05:
Yeah, that probably means it didn't succeed.

Can you share the full output of `./scripts/build-toolchains.sh esp-tools` in a Gist? There's probably a hint in there for why the script failed.

minmin20028, 2022-03-05:
Thanks for your reply.
here`s a link to full log that comes up when running the script.

https://gist.github.com/minmin20028/7015fa90f43718fd3565887ce069ab16

It seems that have to look at the full file because the log contents are not fully included.
Thank you again.

hngenc, 2022-03-06:
Looking through your build log, I can see that a build target called `all-gdb` fails, but I can't seem to figure out why.

You might want to try running `./scripts/build-toolchains.sh esp-tools` on the master branch of Chipyard, and seeing if that works.

minmin20028, 2022-03-12:
It seems like a library dependency problem. But i have no enough time to deal with this issue right now, So I fixed it by setting it up on a new server environment. I`ll let you know later when it`s resolved. Thank you for your help.

minmin20028, 2022-03-12:
I have one more question.

minmin20028, 2022-03-12:
I have one more question.
From what i`ve seen so far, most of the operation in GEMMINI looks proceed like,

(example for convolution) tiled_conv_auto() -> tiled_conv() -> sp_tiled_conv() -> gemmini_loop_ws()

I think, this is a flow that does a lot of preprocessing in software and then finally allocates operations to hardware.

I wonder if there is another flow in GEMMINI system where the software preprocessing is more concise. 
to focus more on measuring the performance of GEMMINI hardware itself not software.

hngenc, 2022-03-12:
We don't have another flow that circumvents the `tiled_conv_auto() -> ...` flow right now, unless you want to call the low-level ISA directly in your own code.

But in either case, the default flow doesn't do any preprocessing on the conv inputs themselves. It only tries to heuristically optimize the tiling factors before offloading work onto Gemmini

closed 2022-04-08 by minmin20028