ucb-bar/gemmini #206: Errors in build Verilator.

Heresyrac, 2022-03-15:
Hi, I am new to gemmini,
I follow the quick start and get some errors among the "Building Gemmini Hardware and Cycle-Accurate Simulators" step.
Here is the error information.
I want to know what is the problem and how to fix it, thank you.

`[info] resolving key references (55373 settings) ...
[info] set current project to chipyardRoot (in build file:/root/chipyard/)
[info] set current project to chipyard (in build file:/root/chipyard/)
[info] compiling 69 Scala sources to /root/chipyard/tools/treadle/target/scala-2.12/classes ...
[error] /root/chipyard/tools/treadle/src/main/scala/treadle/Driver.scala:6:8: object ExecutionOptionsManager is not a member of package firrtl
[error] import firrtl.{AnnotationSeq, ExecutionOptionsManager, HasFirrtlOptions}
[error]        ^
[error] /root/chipyard/tools/treadle/src/main/scala/treadle/Driver.scala:32:20: type ComposableOptions is not a member of package firrtl
[error]     extends firrtl.ComposableOptions {
[error]                    ^
[error] /root/chipyard/tools/treadle/src/main/scala/treadle/Driver.scala:34:41: not found: type ExecutionOptionsManager
[error]   def vcdOutputFileName(optionsManager: ExecutionOptionsManager): String = {
[error]                                         ^
[error] /root/chipyard/tools/treadle/src/main/scala/treadle/Driver.scala:73:9: not found: type ExecutionOptionsManager
[error]   self: ExecutionOptionsManager =>
[error]         ^
[error] /root/chipyard/tools/treadle/src/main/scala/treadle/Driver.scala:77:3: not found: value parser
[error]   parser.note("treadle-options")
[error]   ^
[error] /root/chipyard/tools/treadle/src/main/scala/treadle/Driver.scala:79:3: not found: value parser
[error]   parser
[error]   ^
[error] /root/chipyard/tools/treadle/src/main/scala/treadle/Driver.scala:87:3: not found: value parser
[error]   parser
[error]   ^
[error] /root/chipyard/tools/treadle/src/main/scala/treadle/Driver.scala:95:3: not found: value parser
[error]   parser
[error]   ^
[error] /root/chipyard/tools/treadle/src/main/scala/treadle/Driver.scala:103:3: not found: value parser
[error]   parser
[error]   ^
[error] /root/chipyard/tools/treadle/src/main/scala/treadle/Driver.scala:111:3: not found: value parser
[error]   parser
[error]   ^
[error] /root/chipyard/tools/treadle/src/main/scala/treadle/Driver.scala:119:3: not found: value parser
[error]   parser
[error]   ^
[error] /root/chipyard/tools/treadle/src/main/scala/treadle/Driver.scala:127:3: not found: value parser
[error]   parser
[error]   ^
[error] /root/chipyard/tools/treadle/src/main/scala/treadle/Driver.scala:136:3: not found: value parser
[error]   parser
[error]   ^
[error] /root/chipyard/tools/treadle/src/main/scala/treadle/Driver.scala:144:3: not found: value parser
[error]   parser
[error]   ^
[error] /root/chipyard/tools/treadle/src/main/scala/treadle/Driver.scala:152:3: not found: value parser
[error]   parser
[error]   ^
[error] /root/chipyard/tools/treadle/src/main/scala/treadle/Driver.scala:160:3: not found: value parser
[error]   parser
[error]   ^
[error] /root/chipyard/tools/treadle/src/main/scala/treadle/Driver.scala:168:3: not found: value parser
[error]   parser
[error]   ^
[error] /root/chipyard/tools/treadle/src/main/scala/treadle/Driver.scala:176:3: not found: value parser
[error]   parser
[error]   ^
[error] /root/chipyard/tools/treadle/src/main/scala/treadle/Driver.scala:185:3: not found: value parser
[error]   parser
[error]   ^
[error] /root/chipyard/tools/treadle/src/main/scala/treadle/Driver.scala:206:3: not found: value parser
[error]   parser
[error]   ^
[error] /root/chipyard/tools/treadle/src/main/scala/treadle/Driver.scala:216:3: not found: value parser
[... 97 lines cut ...]
[error]   ^
[error] /root/chipyard/tools/treadle/src/main/scala/treadle/vcd/VCDConfig.scala:48:3: not found: value parser
[error]   parser
[error]   ^
[error] /root/chipyard/tools/treadle/src/main/scala/treadle/vcd/VCDConfig.scala:56:3: not found: value parser
[error]   parser
[error]   ^
[error] /root/chipyard/tools/treadle/src/main/scala/treadle/vcd/VCDConfig.scala:64:3: not found: value parser
[error]   parser
[error]   ^
[error] /root/chipyard/tools/treadle/src/main/scala/treadle/vcd/VCDConfig.scala:73:3: not found: value parser
[error]   parser
[error]   ^
[error] /root/chipyard/tools/treadle/src/main/scala/treadle/vcd/VCDConfig.scala:82:57: no arguments allowed for nullary constructor Object: ()Object
[error] class VCDOptionsManager extends ExecutionOptionsManager("vcd") with HasVCDConfig
[error]                                                         ^
[warn] /root/chipyard/tools/treadle/src/main/scala/treadle/vcd/VCDConfig.scala:5:15: Unused import
[warn] import firrtl.ExecutionOptionsManager
[warn]               ^
[warn] four warnings found
[error] 55 errors found
[error] (treadle / Compile / compileIncremental) Compilation failed
[error] Total time: 9 s, completed Mar 14, 2022, 6:15:05 PM
/root/chipyard/common.mk:105: recipe for target 'generator_temp' failed
make: *** [generator_temp] Error 1`

Heresyrac, 2022-03-16:
> Can you make sure you ran `source env.sh` before building the Verilator simulator?
> 
> Also, which Chipyard commit, and which Gemmini commit are you on?

Yes, I re-run the env.sh and the error information seems same.
The Gemmini commit is : 
commit 87550581d98210540ac013f96fbcb9150be39df7 (HEAD, tag: v0.6.4, origin/master, origin/HEAD)
Author: Hasan Genc <hngenc@berkeley.edu>
Date:   Mon Mar 7 11:27:05 2022 -0800

    Update version number in Quick Setup instructions

Chipyard commit is : 
commit b5d013190d637e634113cb5179f8c8885df1945a (HEAD -> master, tag: 1.5.0, origin/master, origin/HEAD)
Merge: 45684f1b ba1351c8
Author: alonamid <alonamid@eecs.berkeley.edu>
Date:   Mon Jun 14 21:47:41 2021 -0700

    Merge pull request #773 from ucb-bar/dev

    Chipyard 1.5 Release (Dev -> Master Tracking)

hngenc, 2022-03-18:
It seems that you're using an older version of Chipyard.

Can you try running [the commands here](https://github.com/ucb-bar/gemmini#installing-chipyard-and-spike), and verify that they all pass? Those commands should set you up with the correct versions of Gemmini, Chipyard, and Spike.

Heresyrac, 2022-03-24:
@hngenc 
I often encounter network errors in the middle of running these scripts, so I need to run them again many times.  Is it possible that this is the cause of the error？

hngenc, 2022-03-24:
It's possible. Can you share these network errors?

Also, can you share the full output of your `./scripts/init-submodules-no-riscv-tools.sh` and `./scripts/build-toolchains.sh esp-tools` commands?

Heresyrac, 2022-03-24:
@hngenc  
The network error are TLS handshake error and unable to access the google-source, for  using china network. 
I have already fixed them by changing the url to ssh form, and setting proxy.
Here are the outputs：

[Outputs.zip](https://github.com/ucb-bar/gemmini/files/8339695/Outputs.zip)

So, maybe I can fix the error by deleting the chipyard again and repeat those steps without network failure?

hngenc, 2022-03-24:
I don't notice any obvious errors in the outputs you sent me, so I'm not sure why you're getting the error you posted above.

I'm stumped on what could be causing this issue to be honest. I haven't been able to replicate it on my machine.

> So, maybe I can fix the error by deleting the chipyard again and repeat those steps without network failure?

It's worth a try 🤷

Heresyrac, 2022-03-24:
@hngenc 
Now I success to build verilator, but I still encounter errors in building gemini software.

hngenc, 2022-03-24:
Your error says

```
riscv64-unknown-elf-gcc: Command not found
```

This implies that you either:

1. Forgot to run `source env.sh`, or
2. Did not successfully run `./scripts/build-toolchains.sh esp-tools`

Heresyrac, 2022-03-25:
@hngenc Ok I successfully run the test, thank you for the help.

hngenc, 2022-03-25:
No problem!

closed 2022-03-25 by hngenc