ucb-bar/gemmini #207: [Question] difference between pk and linux

hyuns-lee, 2022-03-15:
Hello, I'm trying to use function exp, tanh in math.h.
C code using exp and tanh is well compiled and it is also well executed on linux(through firesim and qemu).
But, if I tried to compile this code and debug with spike, this error message shows while compiling.
```
/chipyard/esp-tools-install/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /tmp/cceVLjf1.o: in function `.L51':
test.c:(.text.startup+0x6de): undefined reference to `__expf_finite'
```
It only shows when the operand of exp() is an array element.
If I use exp(arr[0]), the error above appears but no errors with exp(10).
How can I fix it to use array elements in exp or tanh?

hyuns-lee, 2022-03-15:
I compiled it with the same command in the Makefile of imagenet/.
This it the path.
`chipyard/generators/gemmini/software/gemmini-rocc-tests/imagenet/Makefile`

hngenc, 2022-03-18:
It appears the RISCV baremetal libraries that we use don't implement `exp` or `tanh`.

The reason why `exp(10)` works is that the compiler can calculate that at compile-time, so there's no need to link the math library.

Extending the baremetal math libraries is outside the scope of the Gemmini project's goals, unfortunately. It's possible that future versions of Chipyard will come with more advanced libraries, in which case Gemmini will be able to support these math functions in baremetal mode at runtime.

Until then, I think FireSim might be your best bet.

closed 2022-03-18 by hngenc

hyuns-lee, 2022-03-19:
Okay. Thank you for the information!

hngenc, 2022-03-19:
no problem!