ucb-bar/gemmini #208: Encounter problems with gemmini on FPGA

weihsi, 2022-03-17:
I program gemmini on FPGA, config as follow
```
class WithGemmini(mesh_size: Int, bus_bits: Int) extends Config((site, here, up) => {
  case BuildRoCC => up(BuildRoCC) ++ Seq(
    (p: Parameters) => {
      implicit val q = p
      implicit val v = implicitly[ValName]
      LazyModule(new gemmini.Gemmini(gemmini.GemminiConfigs.defaultConfig.copy(
        meshRows = mesh_size, meshColumns = mesh_size, dma_buswidth = bus_bits)))
    }
  )
  case SystemBusKey => up(SystemBusKey).copy(beatBytes = bus_bits/8)
})
```

```
class RocketBaseConfig extends Config(
  new WithBootROMFile("workspace/bootrom.img") ++
  new WithExtMemSize(0x80000000L) ++
  new WithNExtTopInterrupts(8) ++
  new WithDTS("freechips,rocketchip-vivado", Nil) ++
  new WithDebugSBA ++
  new WithEdgeDataBits(64) ++
  new WithCoherentBusTopology ++
  new WithoutTLMonitors ++
  new BaseConfig)
```
```
class Rocket64b1gem extends Config(
  new WithGemmini(4, 64)  ++
  new WithInclusiveCache  ++
  new WithNBreakpoints(8) ++
  new WithNBigCores(1)    ++
  new WithJtagDTM  ++
  new RocketBaseConfig)
```
and there are some problems I encountered, like I execute gemmini function `tiled_matmul_nn_auto`, and all of the output are zeros. 
I suggest that the reason which causes this problem is `accumulator_memory` can not be recongnize by our linux system.
Is there any way to solve this problem, or I need to change some config to run gemmini normally?
Thanks!

hngenc, 2022-03-18:
We haven't yet tried to put Gemmini on a standalone FPGA.

We typically run:

* Functional simulations on Spike, a simulator which runs on the CPU
* Cycle-accurate simulations on VCS/Verilator
* FPGA-accelerated simulations on FireSim, which runs on the AWS cloud

We can't provide much help on standalone FPGA builds, since we don't have experience with them ourselves. I know that some people outside Berkeley managed to do that, so it should be possible in theory.

closed 2022-03-29 by weihsi

Leo-Z-Li, 2023-12-28:
@weihsi Hi,  I'm facing the same issue, and may I ask if have you finally able to work it out?