ucb-bar/gemmini #212: Simulation Counters of executing onnxruntime-riscv in FireSim

kimjungwow, 2022-04-03:
Hello! 

I am executing image-trainer (onnxruntime-riscv/systolic_runner) in Firesim, and I want to get the simulation counters.
According to [Gemmini_Tutorial](https://berkeley.app.box.com/s/oc16092wrjhijigf065sctt2xosujsdi/file/883910155930), it seems I can get  interesting logs such as memory stats of simulation logs via `memory_stats.csv`.

Referring to [Defining-Custom-Workloads-Tutorial](https://github.com/firesim/firesim/blob/main/docs/Advanced-Usage/Workloads/Defining-Custom-Workloads.rst) and [Running-a-Single-Node-Simulation-Tutorial](https://github.com/firesim/firesim/blob/main/docs/Running-Simulations-Tutorial/Running-a-Single-Node-Simulation.rst)
So, I added the below code in `gemmini-ort.json`, but there is no `memory_stats*.csv` in the output directory. There is only `uartlog` in the output directory, but `uartlog` was there even though I did not add the below code.
```
"common_simulation_outputs": ["uartlog", "memory_stats*.csv"]
```

**What should I do for getting simulation counters such as memory stats in executing onnxruntime-riscv if FireSim?**

For executing image-trainer in FireSim, I referred https://github.com/ucb-bar/gemmini/issues/193#issuecomment-1026504012 and [Running-a-Single-Node-Simulation-Tutorial](https://github.com/firesim/firesim/blob/main/docs/Running-Simulations-Tutorial/Running-a-Single-Node-Simulation.rst).

hngenc, 2022-04-07:
Try these steps:

1. Firesim sometimes creates copies of your json files. Run `find . -name gemmini-ort.json` to find all these copies and make sure that they *all* have `memory_stats*.csv`. Then, run Firesim again to see if you get a memory stats file now.

2. After changing your `gemmini-ort.json`, try running Step 2 [over here](https://github.com/ucb-bar/gemmini/issues/193#issuecomment-1026504012). Make sure to replace the `gemmini-tests.json` part in Step 2 with `gemmini-ort.json`, since that's what you're trying to run. Then, run Firesim again to see if you get a memory stats file now.

3. Make sure that in your `deploy/config_runtime.ini` file, you are running with `workloadname=gemmini-ort.json`, instead of with `workloadname=gemmini-tests.json`.

4. If none of those work, try adding `memory_stats*.csv` to `gemmini-tests.json` instead, and repeat the steps above with `gemmini-tests` instead of `gemmini-ort`. That way, we will at least know whether you can get `memory_stats*.csv` for the default tests. Afterwards, please share your `gemmini-tests.json`, `gemmini-ort.json` and `config_runtime.ini` files in Gists.

kimjungwow, 2022-04-08:
Thank you very much for kind advices! I will try those steps and close this issue or share json/ini files!

closed 2022-04-10 by kimjungwow