ucb-bar/gemmini #214: Tracing memory access in FireSim

kimjungwow, 2022-04-08:
Hello 😁

Sorry for bothering you again with new issue, but I open this issue because I think this issue and my previous issue (#212) are orthogonal.

I want to trace memory access in DRAM and Scratchpad. The example results I want to get are as follows.
```
Cycle,Addr,R/W,...
32,0x1002fd00,Write
36,0x1003ed00,Write
...
33300,0x2df3e00,Read
```

kimjungwow, 2022-04-08:
An issue was created before I finished writing by mistake. Now I have filled out all the questions.

hngenc, 2022-04-09:
Spike should be sufficient for that kind of memory trace.

Try uncommenting this line and rebuilding Spike:

https://github.com/ucb-bar/esp-isa-sim/blob/master/gemmini/gemmini.h#L19

We use that for debugging purposes, but it should also tell you exactly which DRAM and Scratchpad addresses are read from or written to.

kimjungwow, 2022-04-10:
Thank you Hasan! 👍 It is exactly what I looked for.

closed 2022-04-10 by kimjungwow