ucb-bar/gemmini #216: Error running build-verilator.sh

YorkGrizzly, 2022-04-15:
I've tried running the script `./scripts/build-verilator.sh` demonstrated in the tutorial ( with all the relevant environment settings sourced ). I got the following error:

/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.harness.mems.conf -v /scratch/eecs251b-aaf/sp22-project-chen-chang-cai/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.harness.mems.v -f /scratch/eecs251b-aaf/sp22-project-chen-chang-cai/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.harness.mems.fir --mode synflops
[error] Picked up JAVA_TOOL_OPTIONS: -Xmx8G -Xss8M -Djava.io.tmpdir=/scratch/eecs251b-aaf/sp22-project-chen-chang-cai/chipyard/.java_tmp
[success] Total time: 7 s, completed Apr 15, 2022 11:28:15 AM
sort -u /scratch/eecs251b-aaf/sp22-project-chen-chang-cai/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/sim_files.f /scratch/eecs251b-aaf/sp22-project-chen-chang-cai/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/firrtl_black_box_resource_files.top.f /scratch/eecs251b-aaf/sp22-project-chen-chang-cai/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/firrtl_black_box_resource_files.harness.f | grep -v '.*\.\(svh\|h\)$' > /scratch/eecs251b-aaf/sp22-project-chen-chang-cai/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/sim_files.common.f
rm -rf /scratch/eecs251b-aaf/sp22-project-chen-chang-cai/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig
mkdir -p /scratch/eecs251b-aaf/sp22-project-chen-chang-cai/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig
verilator --cc --exe -CFLAGS " -O3 -std=c++11 -I/scratch/eecs251b-aaf/sp22-project-chen-chang-cai/chipyard/esp-tools-install/include -I/scratch/eecs251b-aaf/sp22-project-chen-chang-cai/chipyard/tools/DRAMSim2 -I/scratch/eecs251b-aaf/sp22-project-chen-chang-cai/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig    -D__STDC_FORMAT_MACROS -DTEST_HARNESS=VTestHarness -DVERILATOR -include /scratch/eecs251b-aaf/sp22-project-chen-chang-cai/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.plusArgs -include /scratch/eecs251b-aaf/sp22-project-chen-chang-cai/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/verilator.h" -LDFLAGS " -L/scratch/eecs251b-aaf/sp22-project-chen-chang-cai/chipyard/esp-tools-install/lib -Wl,-rpath,/scratch/eecs251b-aaf/sp22-project-chen-chang-cai/chipyard/esp-tools-install/lib -L/scratch/eecs251b-aaf/sp22-project-chen-chang-cai/chipyard/sims/verilator -L/scratch/eecs251b-aaf/sp22-project-chen-chang-cai/chipyard/tools/DRAMSim2 -lfesvr -ldramsim "   --threads 1 --threads-dpi all -O3 --x-assign fast --x-initial fast --output-split 10000 --output-split-cfuncs 100 --assert -Wno-fatal   +define+PRINTF_COND=\$c\(\"verbose\",\"\&\&\"\,\"done_reset\"\) +define+STOP_COND=\$c\(\"done_reset\"\) --top-module TestHarness --vpi -f /scratch/eecs251b-aaf/sp22-project-chen-chang-cai/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/sim_files.common.f /scratch/eecs251b-aaf/sp22-project-chen-chang-cai/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.top.v /scratch/eecs251b-aaf/sp22-project-chen-chang-cai/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.harness.v /scratch/eecs251b-aaf/sp22-project-chen-chang-cai/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.top.mems.v /scratch/eecs251b-aaf/sp22-project-chen-chang-cai/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.harness.mems.v  -o /scratch/eecs251b-aaf/sp22-project-chen-chang-cai/chipyard/sims/verilator/simulator-chipyard-CustomGemminiSoCConfig -Mdir /scratch/eecs251b-aaf/sp22-project-chen-chang-cai/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig -CFLAGS "-include /scratch/eecs251b-aaf/sp22-project-chen-chang-cai/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig/VTestHarness.h"
%Error: Invalid Option: --threads-dpi
%Error: Command Failed /usr/bin/verilator_bin --cc --exe -CFLAGS \ -O3\ -std\=c\+\+11\ -I/scratch/eecs251b-aaf/sp22-project-chen-chang-cai/chipyard/esp-tools-install/include\ -I/scratch/eecs251b-aaf/sp22-project-chen-chang-cai/chipyard/tools/DRAMSim2\ -I/scratch/eecs251b-aaf/sp22-project-chen-chang-cai/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig\ \ \ \ -D__STDC_FORMAT_MACROS\ -DTEST_HARNESS\=VTestHarness\ -DVERILATOR\ -include\ /scratch/eecs251b-aaf/sp22-project-chen-chang-cai/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.plusArgs\ -include\ /scratch/eecs251b-aaf/sp22-project-chen-chang-cai/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/verilator.h -LDFLAGS \ -L/scratch/eecs251b-aaf/sp22-project-chen-chang-cai/chipyard/esp-tools-install/lib\ -Wl\,-rpath\,/scratch/eecs251b-aaf/sp22-project-chen-chang-cai/chipyard/esp-tools-install/lib\ -L/scratch/eecs251b-aaf/sp22-project-chen-chang-cai/chipyard/sims/verilator\ -L/scratch/eecs251b-aaf/sp22-project-chen-chang-cai/chipyard/tools/DRAMSim2\ -lfesvr\ -ldramsim\  --threads 1 --threads-dpi all -O3 --x-assign fast --x-initial fast --output-split 10000 --output-split-cfuncs 100 --assert -Wno-fatal \+define\+PRINTF_COND\=\$c\(\"verbose\"\,\"\&\&\"\,\"done_reset\"\) \+define\+STOP_COND\=\$c\(\"done_reset\"\) --top-module TestHarness --vpi -f /scratch/eecs251b-aaf/sp22-project-chen-chang-cai/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/sim_files.common.f /scratch/eecs251b-aaf/sp22-project-chen-chang-cai/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.top.v /scratch/eecs251b-aaf/sp22-project-chen-chang-cai/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.harness.v /scratch/eecs251b-aaf/sp22-project-chen-chang-cai/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.top.mems.v /scratch/eecs251b-aaf/sp22-project-chen-chang-cai/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.harness.mems.v -o /scratch/eecs251b-aaf/sp22-project-chen-chang-cai/chipyard/sims/verilator/simulator-chipyard-CustomGemminiSoCConfig -Mdir /scratch/eecs251b-aaf/sp22-project-chen-chang-cai/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig -CFLAGS -include\ /scratch/eecs251b-aaf/sp22-project-chen-chang-cai/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig/VTestHarness.h
make: *** [Makefile:216: /scratch/eecs251b-aaf/sp22-project-chen-chang-cai/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig/VTestHarness.mk] Error 10

I noticed "Invalid Option: --threads-dpi",
so I went ahead and commented out `--threads-dpi all ` on line 108 in the Makefile and got the following when I reran it:

/scratch/eecs251b-aaf/sp22-project-chen-chang-cai/chipyard/sims/verilator/generated-src/chipyard.TestHarness.GemminiRocketConfig/uart.cc:7:0: warning: "_XOPEN_SOURCE" redefined [enabled by default]
 #define _XOPEN_SOURCE
 ^
In file included from /usr/include/stdint.h:25:0,
                 from /usr/lib/gcc/x86_64-redhat-linux/4.8.5/include/stdint.h:9,
                 from /usr/share/verilator/include/verilatedos.h:249,
[... cut ...]
make[1]: *** [<builtin>: verilated.o] Error 1
make[1]: Leaving directory '/scratch/eecs251b-aaf/sp22-project-chen-chang-cai/chipyard/sims/verilator/generated-src/chipyard.TestHarness.GemminiRocketConfig/chipyard.TestHarness.GemminiRocketConfig'
make: *** [Makefile:229: /scratch/eecs251b-aaf/sp22-project-chen-chang-cai/chipyard/sims/verilator/simulator-chipyard-GemminiRocketConfig] Error 2

I was wondering whether I am doing something wrong or there is something I missed.
Any help would be much appreciated, thanks!

hngenc, 2022-04-15:
Your error implies that your Verilator installation may be out-of-date.

Can you check:

1. Which Gemmini commit you are on.

2. Which Chipyard commit you are on.

3. Which Verilator version you are using.

4. Whether you were able to run all the commands in this section successfully?

https://github.com/ucb-bar/gemmini#installing-chipyard-and-spike

YorkGrizzly, 2022-04-18:
I have tried rerunning the steps as listed in the link and is now stuck at `git fetch && git checkout 090e82c473fd28b4eb2011ffcd771ead6076faab`.

1. `git status` shows `HEAD detached at v0.6.4`
2. `git status` shows `HEAD detached at 117624d8`
3. `Verilator 4.028 2020-02-06 rev v4.028`
4. I have tried rerunning the steps as listed in the link and is now stuck at `git fetch && git checkout 090e82c473fd28b4eb2011ffcd771ead6076faab` with error message: `fatal: reference is not a tree: 090e82c473fd28b4eb2011ffcd771ead6076faab`

hngenc, 2022-04-18:
This probably means that one of these two commands failed:

```
./scripts/init-submodules-no-riscv-tools.sh
./scripts/build-toolchains.sh esp-tools
```

Could you share the output of these commands in a Gist?