ucb-bar/gemmini #217: Understand and implement CISC instructions

kimjungwow, 2022-04-25:
Hi! 😀
I have two questions about CISC-type instructions in Gemmini.

1. **How can I implement new CISC instructions such as gemmini_loop_os or gemmini_loop_conv_os?** It seems I can implement gemmini_loop_os or gemmini_loop_conv_os with calling RISC instructions in for loops, but I wonder how to implement new CISC instructions.

2. **How can I understand CISC instructions such as gemmini_loop_ws?** Since CISC instructions may achieve higher throughput than such tiled loops written by non-expert programmers, I want to understand how CISC instructions work in detail. Where is the implementation of CISC instruction corresponding to gemmini_loop_ws?

Thank you!

hngenc, 2022-05-01:
The CISC-like instructions are implemented in `LoopMatmul.scala` and `LoopConv.scala`.

Those files are basically just FSMs which unroll the CISC-like instructions into many smaller RISC-like instructions.

Additionally, the commented out code in `sp_tiled_matmul_ws` and `sp_tiled_conv` correspond exactly to what `LoopMatmul.scala` and `LoopConv.scala` are doing. That might help you to understand what the FSMs do.

kimjungwow, 2022-05-01:
Thank you for your kind explanation! It was a great help to me.

closed 2022-05-01 by kimjungwow