ucb-bar/gemmini #219: Is it correct that it generate so many VTestHarness__*.o/cpp when i use  `./scripts/build-verilator.sh --debug`

hust-gaoyujing, 2022-05-03:
**Problem:** when i use the command_line  `./scripts/build-verilator.sh --debug` (for a simulator that can generate waveforms), it generate so many 
files(VTestHarness__115.o VTestHarness__115.cpp) in the folder(sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.debug), and it seems to continue.  So i want to know whether it's correct.

**Other information:**
`<command-line>: note: this is the location of the previous definition
g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -O3 -std=c++11 -I/home/gaoyujing/chipyard/esp-tools-install/include -I/home/gaoyujing/chipyard/tools/DRAMSim2 -I/home/gaoyujing/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig    -D__STDC_FORMAT_MACROS -DTEST_HARNESS=VTestHarness -DVERILATOR -include /home/gaoyujing/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.plusArgs -include /home/gaoyujing/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/verilator.h -include /home/gaoyujing/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.debug/VTestHarness.h  -DVL_THREADED -std=gnu++14  -c -o VTestHarness__115.o VTestHarness__115.cpp
In file included from /usr/local/share/verilator/include/verilated.h:27,
                 from /usr/local/share/verilator/include/verilated_vcd_c.h:23,
                 from /home/gaoyujing/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/verilator.h:4,
                 from <command-line>:
/usr/local/share/verilator/include/verilatedos.h:252: warning: "__STDC_FORMAT_MACROS" redefined
  252 | #define __STDC_FORMAT_MACROS
      | 
<command-line>: note: this is the location of the previous definition
g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -O3 -std=c++11 -I/home/gaoyujing/chipyard/esp-tools-install/include -I/home/gaoyujing/chipyard/tools/DRAMSim2 -I/home/gaoyujing/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig    -D__STDC_FORMAT_MACROS -DTEST_HARNESS=VTestHarness -DVERILATOR -include /home/gaoyujing/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.plusArgs -include /home/gaoyujing/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/verilator.h -include /home/gaoyujing/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.debug/VTestHarness.h  -DVL_THREADED -std=gnu++14  -c -o VTestHarness__116.o VTestHarness__116.cpp
In file included from /usr/local/share/verilator/include/verilated.h:27,
                 from /usr/local/share/verilator/include/verilated_vcd_c.h:23,
                 from /home/gaoyujing/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/verilator.h:4,
                 from <command-line>:
/usr/local/share/verilator/include/verilatedos.h:252: warning: "__STDC_FORMAT_MACROS" redefined
  252 | #define __STDC_FORMAT_MACROS
      | 
<command-line>: note: this is the location of the previous definition`

hngenc, 2022-05-03:
That's expected.

VCS takes less time to build, so I would recommend that if you have a license.

hust-gaoyujing, 2022-05-03:
Thank you! It works , but it really takes me a long time.

hngenc, 2022-05-03:
What if you try adding a `-j8` to this line:

https://github.com/ucb-bar/gemmini/blob/master/scripts/build-verilator.sh#L32

That should make your build faster.

(In fact, I should probably make that the default).

hust-gaoyujing, 2022-05-03:
I will be try it later.