ucb-bar/gemmini #220: error when run build-verilator.sh and run-verilator sh

kahretsinanil, 2022-05-09:


hngenc, 2022-05-10:
1. Which Gemmini commit and Chipyard commit are you on?
2. Can you confirm that you ran all the instructions [here](https://github.com/ucb-bar/gemmini#installing-chipyard-and-spike) and [here](https://github.com/ucb-bar/gemmini#setting-up-gemmini) correctly?

kahretsinanil, 2022-05-10:
> 1. Which Gemmini commit and Chipyard commit are you on?
> 2. Can you confirm that you ran all the instructions [here](https://github.com/ucb-bar/gemmini#installing-chipyard-and-spike) and [here](https://github.com/ucb-bar/gemmini#setting-up-gemmini) correctly?

Hi,
1)
git status shows HEAD detached at v0.6.4
git status shows HEAD detached at 117624d8

2)
Yes, i did it. There is no problem in ```./scripts/build-spike.sh``` and ```./scripts/run-spike.sh resnet50```
Bu after run ```./scripts/build-verilator.sh```, the ```verilator``` file in ```generated-src/``` folder is empty.

I also tried to add ```-debug``` from https://github.com/ucb-bar/gemmini/issues/196, but i couldn't follow where should i add it.

hngenc, 2022-05-10:
You can add `--debug` to `./scripts/build-verilator.sh` and to `./scripts/run-verilator.sh`, but I don't think they will fix your problem. They just allow your Verilator simulations to produce VCD waveform files.

I'll try to replicate the issue you're seeing. I'm not sure what could be going wrong.

hngenc, 2022-05-10:
One more thing:

1. Can you share the full output of your `./build-verilator.sh` command in a Gist?

2. Can you tell me what files you see in `chipyard/sims/verilator`?

kahretsinanil, 2022-05-10:
> 

I've already added  ```--debug``` but nothing changed.

But when I try to run ``` ./scripts/run-verilator.sh template```, the output becomes 
``` ./scripts/run-verilator.sh: line 91: ./simulator-chipyard-CustomGemminiSoCConfig: No such file or directory```

On the other hand if I try to run ```./scripts/run-verilator.sh --debug template```, the output becomes ```./scripts/run-verilator.sh: line 91: ./simulator-chipyard-CustomGemminiSoCConfig-debug: No such file or directory``` so there is additional ```-debug``` in here but the problem is similar.

output of the ```./build-verilator.sh``` is here:
https://gist.github.com/kahretsinanil/e9fa5811d6de0e0862301d000c7f7f78

inside of the ```chipyard/sims/verilator``` is here:
https://gist.github.com/kahretsinanil/9c03fc93be3be7226e9c98e74f27e4b3

hngenc, 2022-05-11:
Can you try this as well?

```
cd chipyard/sims/verilator
make CONFIG=GemminiRocketConfig
ls
```

And send me the output of that?

kahretsinanil, 2022-05-11:
> Can you try this as well?
> 
> ```
> cd chipyard/sims/verilator
> make CONFIG=GemminiRocketConfig
> ls
> ```
> 
> And send me the output of that?

Hi,
Here it is. Is there any missing files in there?
https://gist.github.com/kahretsinanil/8fcc8adc8141ae7655328ad33d484ac1

hngenc, 2022-05-12:
You should have seen a file called `simulator-chipyard-GemminiRocketConfig`. I haven't been able to replicate your issue, but I'm still trying. I suspect that your Chipyard installation wasn't setup properly.

Can you try deleting your Chipyard/Gemmini installation and rerunning the commands [here](https://github.com/ucb-bar/gemmini#quick-start)? However, this time, change these two lines:

```
./scripts/init-submodules-no-riscv-tools.sh
./scripts/build-toolchains.sh esp-tools
```

to

```
./scripts/init-submodules-no-riscv-tools.sh
echo $? >init-submodules-no-riscv-tools-success.txt
./scripts/build-toolchains.sh esp-tools
echo $? >build-toolchains-success.txt
```

That will allow us to make sure this time that those commands actually succeeded. I've seen them fail in the past without the user noticing, because they don't always print out very obvious error messages.

closed 2022-06-19 by kahretsinanil