ucb-bar/gemmini #221: where can i find the simulation logs?

hust-gaoyujing, 2022-05-18:
After finishing the simulation(run-verilator.sh) ,where can i find the simulation logs that record the results of simulation and the counters of SOC? Or whether it have simulation logs?

hngenc, 2022-05-26:
We discuss this a little bit in our [IISWC tutorial](https://sites.google.com/berkeley.edu/gemminitutorialiiswc2021/). (We will also have a new, more up-to-date tutorial up in a few months).

You can spit out results from the SoC counters by using `printf` as in [this example program](https://github.com/ucb-bar/gemmini-rocc-tests/blob/e326e7c43457ff08669fe88edcaa395d846474d8/bareMetalC/gemmini_counter.c). The results will be printed onto your screen as you run `run-verilator.sh`.

If you want to use Firesim counters (which won't appear on an SoC that you tapeout), then make sure to set [this config parameter](https://github.com/ucb-bar/gemmini/blob/master/src/main/scala/gemmini/GemminiConfigs.scala#L91) to `true` on your Firesim instance. You will also need to enable Firesim's ["printf synthesis"](https://docs.fires.im/en/stable/Advanced-Usage/Debugging-and-Profiling-on-FPGA/Printf-Synthesis.html) and ["AutoCounter"](https://docs.fires.im/en/stable/Advanced-Usage/Debugging-and-Profiling-on-FPGA/AutoCounter.html) features as described in the links.

Afterwards, you should be able to see the Firesim counter and printf results under `firesim/deploy/results-workload` in your manager instance.

Let me know if you have any questions about any of that!

hust-gaoyujing, 2022-05-26:
Thanks.You explained it very well.