ucb-bar/gemmini #223: gemmini waveform variable question again

minmin20028, 2022-05-28:
Hi. 
I closed the issue #222  because I thought it was resolved after asking a question.
But, As I continue to work, a variable may or may not be added, I cannot figure out why.

So, I would like to know if there is anything to know about the process of registering a variable in waveform.

Always thanks for your work.

hngenc, 2022-05-28:
Do you mean that a variable doesn't appear in your VCD/VPD waveform?

If so, it's probably just been optimized out by Chisel. You can prevent that from happening by using Chisel's ["dontTouch" annotation](https://www.chisel-lang.org/api/latest/chisel3/dontTouch$.html).

You can also see an example of Gemmini using the `dontTouch` annotation [in our own source code](https://github.com/ucb-bar/gemmini/blob/87550581d98210540ac013f96fbcb9150be39df7/src/main/scala/gemmini/ReservationStation.scala#L78).

minmin20028, 2022-05-28:
That`s correct. Thanks!!

hngenc, 2022-05-28:
No problem!

closed 2022-05-28 by hngenc