ucb-bar/gemmini #225: GEMMINI Implementation on Xilinx demonstrator

deykid, 2022-06-27:
Hi,

I would like to ask some questions about **GEMMINI** interfaces.
I did the installation of the GEMMINI project and I ran with Vivado the file: *chipyard/generators/gemmini/generated-src/verilator/chipyard.TestHarness.CustomGemminiSoCConfig.top.v*
I guess this is the top of GEMMINI and I took a look at the block design.
I see different interfaces (JTAG, AXI4, Serial TL, UART): [design_1.pdf](https://github.com/ucb-bar/gemmini/files/8990241/design_1.pdf)

My questions are as follows:

- What's the role of the **AXI4** and **Serial TL**?
   - I assume that the AXI4 is used to connect to memory, but which one? DDR?
     - If it is used to connect to DDR, then is the physical address range fixed or can it be configured?

jerryz123, 2022-07-01:
AXI connects to whatever is modeling DRAM. 
The physical address base should always be 0x8000_0000. But the size can be configured. 

Serial tl is a debut interface that is used when the simulated device boots up.

hngenc, 2022-07-01:
This question would probably be best for the [Chipyard mailing list](https://groups.google.com/g/chipyard)

closed 2022-07-01 by hngenc

deykid, 2022-07-04:
> AXI connects to whatever is modeling DRAM. The physical address base should always be 0x8000_0000. But the size can be configured.
> 
> Serial tl is a debut interface that is used when the simulated device boots up.

> AXI connects to whatever is modeling DRAM. The physical address base should always be 0x8000_0000. But the size can be configured.
> 
> Serial tl is a debut interface that is used when the simulated device boots up.

Thank you for your quick answer ! :D