ucb-bar/gemmini #228: Error during build-verilator

DrSpace, 2022-07-19:
There is an error during build-verilator. But it still goes to the very end and seems to work well. Can someone explain this?

[image]

hngenc, 2022-07-19:
Just ignore that "error". It seems to be an issue with Chipyard, but it doesn't affect any of the simulations.

deykid, 2022-07-27:
I got the same error, but it didn't impact my simulation or anything else.

xlgforever, 2022-10-26:
I got the same error,but  the simulation failed:
```
stHarness.CustomGemminiSoCConfig.harness.mems.conf -v /home/hao/wrk/git/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.harness.mems.v -f /home/hao/wrk/git/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.harness.mems.fir --mode synflops
[error] Picked up JAVA_TOOL_OPTIONS: -Xmx8G -Xss8M -Djava.io.tmpdir=/home/hao/wrk/git/chipyard/.java_tmp
[success] Total time: 4 s, completed Oct 26, 2022, 11:44:06 AM
sort -u /home/hao/wrk/git/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/sim_files.f /home/hao/wrk/git/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/firrtl_black_box_resource_files.top.f /home/hao/wrk/git/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/firrtl_black_box_resource_files.harness.f | grep -v '.*\.\(svh\|h\)$' > /home/hao/wrk/git/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/sim_files.common.f
rm -rf /home/hao/wrk/git/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig
mkdir -p /home/hao/wrk/git/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig
verilator --cc --exe -CFLAGS " -O3 -std=c++11 -I/home/hao/wrk/git/chipyard/esp-tools-install/include -I/home/hao/wrk/git/chipyard/tools/DRAMSim2 -I/home/hao/wrk/git/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig    -D__STDC_FORMAT_MACROS -DTEST_HARNESS=VTestHarness -DVERILATOR -include /home/hao/wrk/git/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.plusArgs -include /home/hao/wrk/git/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/verilator.h" -LDFLAGS " -L/home/hao/wrk/git/chipyard/esp-tools-install/lib -Wl,-rpath,/home/hao/wrk/git/chipyard/esp-tools-install/lib -L/home/hao/wrk/git/chipyard/sims/verilator -L/home/hao/wrk/git/chipyard/tools/DRAMSim2 -lfesvr -ldramsim "   --threads 1 --threads-dpi all -O3 --x-assign fast --x-initial fast --output-split 10000 --output-split-cfuncs 100 --assert -Wno-fatal   +define+PRINTF_COND=\$c\(\"verbose\",\"\&\&\"\,\"done_reset\"\) +define+STOP_COND=\$c\(\"done_reset\"\) --top-module TestHarness --vpi -f /home/hao/wrk/git/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/sim_files.common.f /home/hao/wrk/git/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.top.v /home/hao/wrk/git/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.harness.v /home/hao/wrk/git/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.top.mems.v /home/hao/wrk/git/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.harness.mems.v  -o /home/hao/wrk/git/chipyard/sims/verilator/simulator-chipyard-CustomGemminiSoCConfig -Mdir /home/hao/wrk/git/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig -CFLAGS "-include /home/hao/wrk/git/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig/VTestHarness.h"
%Error: Invalid Option: --threads-dpi
%Error: Command Failed /usr/bin/verilator_bin --cc --exe -CFLAGS ' -O3 -std=c++11 -I/home/hao/wrk/git/chipyard/esp-tools-install/include -I/home/hao/wrk/git/chipyard/tools/DRAMSim2 -I/home/hao/wrk/git/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig    -D__STDC_FORMAT_MACROS -DTEST_HARNESS=VTestHarness -DVERILATOR -include /home/hao/wrk/git/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.plusArgs -include /home/hao/wrk/git/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/verilator.h' -LDFLAGS ' -L/home/hao/wrk/git/chipyard/esp-tools-install/lib -Wl,-rpath,/home/hao/wrk/git/chipyard/esp-tools-install/lib -L/home/hao/wrk/git/chipyard/sims/verilator -L/home/hao/wrk/git/chipyard/tools/DRAMSim2 -lfesvr -ldramsim ' --threads 1 --threads-dpi all -O3 --x-assign fast --x-initial fast --output-split 10000 --output-split-cfuncs 100 --assert -Wno-fatal '+define+PRINTF_COND=$c("verbose","&&","done_reset")' '+define+STOP_COND=$c("done_reset")' --top-module TestHarness --vpi -f /home/hao/wrk/git/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/sim_files.common.f /home/hao/wrk/git/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.top.v /home/hao/wrk/git/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.harness.v /home/hao/wrk/git/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.top.mems.v /home/hao/wrk/git/chipyard/sims/verilat [... cut ...] erated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig/VTestHarness.h'
Makefile:214: recipe for target '/home/hao/wrk/git/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig/VTestHarness.mk' failed
make: *** [/home/hao/wrk/git/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig/VTestHarness.mk] Error 10
```

closed 2022-11-23 by hngenc